DynamicDevices/meta-mono #11: Error compiling 5.18.0.268 for ARM

cmcqueen, 2019-07-22:
When I try to compile version 5.18.0.268 for an ARM-based machine, I get an error:

    ibmono_system_native_la-pal_errno.o: file not recognized: File format not recognized

This seems to relate to [Can't cross-compile 5.16.0.179: libmono_system_native built on the wrong platform · Issue #11744 · mono/mono](https://github.com/mono/mono/issues/11744).

This also affects version 5.16.0.220, but not 5.14.0.177 or 5.18.1.0.

cmcqueen, 2019-07-25:
I'm trying the following in `mono_5.18.%.bbappend` file to try to fix this issue:

    # The source package erroneously contains some build output files from a
    # build for x86. These need to be deleted.
    
    remove_build_output_files() {
        find ${S} \( -name "*.o" -o -name "*.lo" \) -exec rm -f {} \;
    }
    
    python do_unpack_append() {
        bb.build.exec_func('remove_build_output_files', d)
    }

directhex, 2019-07-30:
5.18.0.271+.

ajlennon, 2019-07-31:
Is this still an issue? Want to give me a PR? Thanks.

ajlennon, 2020-02-09:
Closing as no response

bith3ad, 2021-06-17:
At least with 5.18.0.240 it is still a problem

bith3ad, 2021-06-17:
Okay, after checking the code I found the issue. The globing mechanism you are using seems like a bash globbing. Bitbake uses /bin/sh and not /bin/bash and so this depends on the host. This don't work least on debian/ubunute where the default is the dash. So we need to fix the globbing. I prepared a patch which I can send if you want.

ajlennon, 2021-06-17:
Please

ajlennon, 2022-02-07:
No activity for some time so closing

closed 2022-02-07 by ajlennon