DynamicDevices/meta-mono #12: Failed to build: 'mono-native' ... files already exist

befedo, 2019-09-11:
Hi there,

I'm trying to build the *mono-helloworld* receipe and have added *meta-mono* to my bblayers.conf. After running `bitbake mono-helloworld` the build fails with:
```{text}
ERROR: The recipe mono-native is trying to install files into a shared area when those files already exist.
```

I've also found a mail on the yocto [mailing list](https://lists.yoctoproject.org/pipermail/yocto/2018-April/041010.html) but there's no response.

How could I figure out what went wrong?

Thanks for your time and best regards,
Befedo

ajlennon, 2019-09-11:
Hi - what version of meta-mono and what version of Yocto are you using?

befedo, 2019-10-29:
Hi @ajlennon,

Thanks for your response, and sorry for the delay.

I'm on Ubuntu 16.04LTS using the latest github version of meta-mono. Some version info:
```{text}
$ bitbake -e virtual/kernel | grep "^PV"
   PV="4.1.8+gitAUTOINC+cf9ab7112d_dbe692d91c"
$ bitbake --version
   BitBake Build Tool Core version 1.28.0
$ git rev-parse --short HEAD
   c4adf80
```

`poky` config looks like:
```{text}
meta-yocto/conf/distro/poky.conf
  DISTRO_VERSION = "2.0.3"
  DISTRO_CODENAME = "jethro"
```

Thanks for your time and best regards,
Befedo

ajlennon, 2020-02-09:
Is this still an issue?

sheraz045, 2020-02-18:
yes i am also facing similar issue what is the solution for it

ajlennon, 2020-02-18:
All the builds are completing fine at present for Zeus, Warrior and Thud. I'm not supporting older releases (in line with the current Yocto project)

@see: https://wiki.yoctoproject.org/wiki/Releases

sheraz045, 2020-02-18:
I am working imx6ull  and yocto version is given below 

DISTRO = "poky"
DISTRO_NAME = "Poky (Yocto Project Reference Distro)"
DISTRO_VERSION = "2.1.1"
DISTRO_CODENAME = "krogoth"
SDK_VENDOR = "-pokysdk"
SDK_VERSION := "${@'${DISTRO_VERSION}'.replace('snapshot-${DATE}','snapshot')}"

FILE :- sources/poky/meta-poky/conf/distro/poky.conf

ajlennon, 2020-02-18:
Ah so it's Krogoth - that's also not supported. Have you tried the Krogoth branch of meta-mono?

sheraz045, 2020-02-18:
No can you provide me the link for krogoth?

ajlennon, 2020-02-18:
https://github.com/DynamicDevices/meta-mono/tree/krogoth

sheraz045, 2020-02-18:
can you also provide me link for sumo?

ajlennon, 2020-02-18:
You just pull downt the branch box on the top left 

https://github.com/DynamicDevices/meta-mono/tree/sumo

sheraz045, 2020-02-18:
we were able to build it , but running the application gives the following Warning/Error

X does not support your locale
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
System.NotSupportedException: [X11] GLX extension is not supported.

sheraz045, 2020-02-25:
How is x11 and mono connected, does mono application require additional x11 configuration .
Any link or suggestion would be appreciated.

ajlennon, 2021-02-03:
The oldest branch we test and support is sumo. Closing this issue - feel free to reopen

see: http://jenkins.dynamicdevices.co.uk:8080/job/yocto/

closed 2021-02-03 by ajlennon