DynamicDevices/meta-mono #15: QA Issue: /usr/lib/mono/4.5/Microsoft.CodeAnalysis.Scripting.dll contained in package mono-libs-4.5 requires mono(System.Runtime.Loader), but no providers found in RDEPENDS_mono-libs-4.5

sheraz045, 2020-01-13:
ERROR: mono-5.14.0.177-r0 do_package_qa: QA Issue: /usr/lib/mono/4.5/Microsoft.CodeAnalysis.Scripting.dll contained in package mono-libs-4.5 requires mono(System.Runtime.Loader), but no providers found in RDEPENDS_mono-libs-4.5? [file-rdeps]
ERROR: mono-5.14.0.177-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: mono-5.14.0.177-r0 do_package_qa: Function failed: do_package_qa

I'm on Ubuntu 16.04LTS

$ bitbake -e virtual/kernel | grep "^PV="
PV="4.14.98"

$ bitbake --version
BitBake Build Tool Core version 1.38.0

meta-mono            = "master:1a54dbe8027644e68b02f4e44ca2f631d7e80e99"

ajlennon, 2020-02-09:
Is this still an issue?

ajlennon, 2020-02-18:
No response so closing issue

closed 2020-02-18 by ajlennon