DynamicDevices/meta-mono #19: What is needed to support Winforms with opacity?

befedo, 2020-01-30:
Hi there,

I've got `x11-base` in `IMAGE_FEATURES` and added `mono`, `libgdiplus` and `gtk-sharp` to `IMAGE_INSTALL` but when running a Form which has an opacity attribute it seems to be ignored and shows up like a white coloured area.

I'm sure I've done something wrong thus this question.

Thanks for your time, and best regards,
Befedo

ajlennon, 2020-01-30:
At the minute I'm finding my basic WinForms tests fail on master so need to figure out where that regression has occurred and then we can see :) 

I would like to see a test case for this if you can create one?

ajlennon, 2020-01-30:
ref: https://github.com/DynamicDevices/meta-mono/blob/master/lib/oeqa/runtime/cases/mono.py

befedo, 2020-01-30:
@ajlennon Thanks for the link, I'm looking into it and try to create a test. But I'm not sure if I'm capable ...

My knowledge about the Openembedded ecosystem isn't very huge.

ajlennon, 2020-01-30:
Sure thing. Understood. Basically we need a new function in the form here

https://github.com/DynamicDevices/meta-mono/blob/master/lib/oeqa/runtime/cases/mono.py#L38

That can compile up a source file you put in files to do your test. Dunno how you'll test the background colour programmatically. That might take some jiggery grabbing pixels

ajlennon, 2020-02-11:
I found this @befedo 

> How can I make my windows alpha blended? (transparent)#
>Mono’s Winform implementation supports transparency on its windows as long as the underlying windowing system has support for it.

>For Unix/X11 users this means that they must have the COMPOSITE extension enabled on their server, and they must be running a compositing manager, like xcompmgr.

>The GenToo Linux Wiki has a good description on how to setup the Xorg server for transparency support.

https://www.mono-project.com/docs/faq/winforms/

befedo, 2020-02-11:
@ajlennon Thanks for the hint! I’ll try it out and let you know ... 👍🏻

ajlennon, 2022-02-07:
No activity for some time so closing

closed 2022-02-07 by ajlennon