DynamicDevices/meta-mono #38: msbuild recipe broken

splitice, 2020-03-20:
```
| ** Downloading MSBUILD from https://github.com/Microsoft/msbuild/releases/download/mono-hosted-msbuild-v0.03/mono_msbuild_d25dd923839404bd64cc63f420e75acf96fc75c4.zip
| + curl -sL -o /yocto/resin-board/build/tmp/work/x86_64-linux/msbuild-native/15.4-r0/git/packages/msbuild.zip https://github.com/Microsoft/msbuild/releases/download/mono-hosted-msbuild-v0.03/mono_msbuild_d25dd923839404bd64cc63f420e75acf96fc75c4.zip
| WARNING: exit code 77 from a shell command.
```

Tested on an otherwise empty (base) image.

ajlennon, 2020-03-20:
Yeah I spotted that. Would be good to have a fix

AnandShastry, 2020-09-02:
Its seems to be a permission issue. tried running cibuild.sh with sudo, this step will pass.. But for sure there are comparability issues, because of which build fails.

PR #53 (2021-01-22): Add msbuild 16.6
The current msbuild 15.4 recipe is broken. This pull request adds a new recipe for version 16.6. I was only able to test the msbuild-native recipe with mono 6.8.0.105 and poky zeus, every other configuration is completely untested.

I added a recipe for dotnet as it is a dependency of msbuild. I copied it from [intel-iot-devkit/meta-iot-cloud](https://github.com/intel-iot-devkit/meta-iot-cloud) as I didn't want to depend on the whole layer just for a single recipe. The layer is MIT licensed, this probably needs attribution.

There is also a change to use correct paths in the c [... cut]
closed 2021-01-22 by ajlennon