DynamicDevices/meta-mono #40: cpu utilization

sheraz045, 2020-04-08:
Hi,
   I have created a mono application by taking mono-compiler from the krogoth branch
while running the application
I run the application using the following command 

mono Application.exe &

PID | USER | PR | NI | VIRT   | RES     | SHR    | S %CPU | %MEM | TIME+ | COMMAND
**607 | root  | 20 | 0 | 167040 | 87664 | 34316 | R 94.1    | 17.       | 4:12.68 | mono**
628 | root  | 20 | 0 | 30776   | 20120 | 16476 | S 4.3      | 4.0       | 0:12.00 | Xwayland
642 | root  | 20 | 0 | 3004     | 1840   | 1544   | R 1.0      | 0.4       | 0:01.66 | top
520 | root  | 20 | 0 | 50776   | 8864   | 7124   | .8 0.7     | 2.         | 0:04.00 | weston
1     | root  | 20 | 0 | 1708     | 328     | 292     | S 0.0      | 0.1       | 0:01.56 | init
2     | root  | 20 | 0 | 0           | 0         | 0         | S 0.0      | 0.0       | 0:00.00 | kthreadd

top is giving 94.1 % CPU utilization
how can i reduce this CPU utilization

ajlennon, 2020-04-08:
Hi @sheraz045 that is interesting. It sounds like more of a generic Mono problem. Can  you provide a simple source code example?

sheraz045, 2020-04-08:
I am trying to run the following application https://github.com/farmerbriantee/AgOpenGPS version 3.0.3 on the krogth branch of yocto on imx6ull board

ajlennon, 2020-04-08:
Have you run it on any other builds of Mono (e.g. desktop) without issue?

sheraz045, 2020-04-08:
yes i had ran it ubuuntu 16.04 , There the application was running without lag, but in arm board i am facing lag while the application is running along with high CPU utilization.

ajlennon, 2020-04-08:
Hi - it's not something I can look at now. This sounds like it might be worth you asking on the main Mono lists.

closed 2020-04-29 by sheraz045