DynamicDevices/meta-mono #41: 6.8.0.105 native toolchain `mcs` unhandled exception

cmcqueen, 2020-05-13:
I'm using Yocto sumo, targetting an ARM target (BeagleBone Black based system). I've built `mono-native` and `mono` version 6.8.0.105.

I have a recipe that uses `mcs` from `mono-native` to compile a `.cs` file. But `mcs` fails with an exception:

```
Unhandled Exception:
System.TypeInitializationException: The type initializer for 'System.Console' threw an exception. ---> System.TypeInitializationException: The type initializer for 'System.ConsoleDriver' threw an exception. ---> System.TypeInitializationException: The type initializer for 'Sys' threw an exception. ---> System.DllNotFoundException: System.Native assembly:<unknown assembly> type:<unknown type> member:(null)
  at (wrapper managed-to-native) Interop+Sys.LChflagsCanSetHiddenFlag()
  at Interop+Sys..cctor () [0x00000] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
   --- End of inner exception stack trace ---
  at System.IO.FileSystem.FileExists (System.ReadOnlySpan`1[T] fullPath, System.Int32 fileType, Interop+ErrorInfo& errorInfo) [0x00007] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
  at System.IO.FileSystem.DirectoryExists (System.ReadOnlySpan`1[T] fullPath, Interop+ErrorInfo& errorInfo) [0x00000] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
  at System.IO.FileSystem.DirectoryExists (System.ReadOnlySpan`1[T] fullPath) [0x00000] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
  at System.IO.Directory.Exists (System.String path) [0x0001e] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
  at System.TermInfoDriver.SearchTerminfo (System.String term) [0x00044] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
  at System.TermInfoDriver..ctor (System.String term) [0x0004b] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
  at System.ConsoleDriver.CreateTermInfoDriver (System.String term) [0x00000] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
  at System.ConsoleDriver..cctor () [0x0004d] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
   --- End of inner exception stack trace ---
  at System.Console.SetupStreams (System.Text.Encoding inputEncoding, System.Text.Encoding outputEncoding) [0x00007] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
  at System.Console..cctor () [0x0007d] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
   --- End of inner exception stack trace ---
  at Mono.CSharp.Driver.Main (System.String[] args) [0x00019] in <8aa40ac63bf14e279634f3d008a836c5>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.TypeInitializationException: The type initializer for 'System.Console' threw an exception. ---> System.TypeInitializationException: The type initializer for 'System.ConsoleDriver' threw an exception. ---> System.TypeInitializationException: The type initializer for 'Sys' threw an exception. ---> System.DllNotFoundException: System.Native assembly:<unknown assembly> type:<unknown type> member:(null)
  at (wrapper managed-to-native) Interop+Sys.LChflagsCanSetHiddenFlag()
  at Interop+Sys..cctor () [0x00000] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
   --- End of inner exception stack trace ---
  at System.IO.FileSystem.FileExists (System.ReadOnlySpan`1[T] fullPath, System.Int32 fileType, Interop+ErrorInfo& errorInfo) [0x00007] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
  at System.IO.FileSystem.DirectoryExists (System.ReadOnlySpan`1[T] fullPath, Interop+ErrorInfo& errorInfo) [0x00000] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
  at System.IO.FileSystem.DirectoryExists (System.ReadOnlySpan`1[T] fullPath) [0x00000] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
  at System.IO.Directory.Exists (System.String path) [0x0001e] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
  at System.TermInfoDriver.SearchTerminfo (System.String term) [0x00044] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
  at System.TermInfoDriver..ctor (System.String term) [0x0004b] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
  at System.ConsoleDriver.CreateTermInfoDriver (System.String term) [0x00000] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
  at System.ConsoleDriver..cctor () [0x0004d] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
   --- End of inner exception stack trace ---
  at System.Console.SetupStreams (System.Text.Encoding inputEncoding, System.Text.Encoding outputEncoding) [0x00007] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
  at System.Console..cctor () [0x0007d] in <3175dabcd4ee435ea3015c7f8675c3de>:0 
   --- End of inner exception stack trace ---
  at Mono.CSharp.Driver.Main (System.String[] args) [0x00019] in <8aa40ac63bf14e279634f3d008a836c5>:0 
```

cmcqueen, 2020-05-13:
Is this maybe related to [mono/mono/issues/16991](/mono/mono/issues/16991)?

cmcqueen, 2020-05-21:
Is anyone else having this issue?

I'm currently using Yocto sumo. Is it likely that this issue could be resolved by upgrading to a newer Yocto version?

ajlennon, 2020-05-21:
Can you submit a simple test case?

cmcqueen, 2020-05-22:
> Can you submit a simple test case?

I tried `mono-helloworld` as a simple test case. That worked fine, which led me to question my original report. I found that the recipe built okay when `inherit mono` was added. For some reason the recipe still worked in the past without `inherit mono`.

Anyway, I think that means this issue can be closed. Thanks for your help.

closed 2020-05-22 by cmcqueen

ajlennon, 2020-05-22:
ah ok - great news @cmcqueen :)