DynamicDevices/meta-mono #44: How to deploy libgluezilla?

befedo, 2020-06-11:
Hi there,

Receiving the output:
```{text}
libgluezilla not found. To have webbrowser support, you need libgluezilla installed
```
How does one deploy `libgluezilla` to the target?

Thanks for your time and best regards,
Befedo

ajlennon, 2020-06-11:
Excellent question. I had a quick google and I see there's some confusion about this. Not sure I'm afraid :-/

ajlennon, 2020-06-11:
Can you make it work on a standard Ubuntu install ?

befedo, 2020-06-11:
Hey @ajlennon,

Thanks for the quick reply!

We do some async picture loading like:
```{C#}
pictureBoxMapTile.WaitOnLoad = false;
pictureBoxMapTile.LoadAsync(urlStr);
pictureBoxMapTile.LoadCompleted += _LoadCompleted;
```
this evolves to `System.Net.WebClient.DownloadDataAsync` and crashes with:
```{text}
[ERROR] FATAL UNHANDLED EXCEPTION: System.NotSupportedException: WebClient does not support concurrent I/O operations.
```
the same exception occurs under plain arch linux:
```{text}
$ uname -a
Linux *** 5.6.15-arch1-1 #1 SMP PREEMPT Wed, 27 May 2020 23:42:26 +0000 x86_64 GNU/Linux
$ mono --version
Mono JIT compiler version 6.4.0 (makepkg/fe64a4765e6 Sat 16 May 2020 10:04:29 AM UTC)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
	TLS:           __thread
	SIGSEGV:       altstack
	Notifications: epoll
	Architecture:  amd64
	Disabled:      none
	Misc:          softdebug 
	Interpreter:   yes
	LLVM:          supported, not enabled.
	Suspend:       hybrid
	GC:            sgen (concurrent by default)
```

Now i'm afraid too ...

ajlennon, 2020-06-11:
Yeah. Humm. Maybe first off can you create a really simple test case to exhibit this and  then try getting it working on a full-fat distro like Ubuntu or something?

ajlennon, 2020-06-11:
Er - humm again - just a sec - that looks more like something that's just not supported in the WebControl rather than the original library problem?

befedo, 2020-06-26:
Hi @ajlennon,

Sorry for the noise and the delay, we figured out that there was a misconception in our `Picturebox` implementation. Using one instance twice for async load, which isn't supported.

The `libgluezilla` dependency remains, but I think we should switch to `Mono.Browser` for that.

Thanks for your time and best regards,
Befedo

closed 2020-06-26 by befedo