DynamicDevices/meta-mono #46: [Dunfell] Cannot build mono

chaitu236, 2020-08-20:
mono-6.10.0.104 isn't building with dunfell and fails with error

`./include/private/gc_atomic_ops.h:105:11: fatal error: atomic_ops.h: No such file or directory
`

Ref: http://jenkins.dynamicdevices.co.uk:8080/job/meta-mono%20(dunfell,x86-64)/10/console

AnandShastry, 2020-08-27:
Its failing for me also with the same error. @ajlennon , Please suggest.

cmcqueen, 2020-09-07:
I'm seeing this error also in Yocto warrior.

cmcqueen, 2020-09-07:
It looks as though this is an upstream issue — see mono/mono#20299.

zboszor, 2021-01-26:
Please try the current master.

ajlennon, 2021-02-03:
The latest CI builds of master build and test dunfell successfully so am closing this

http://jenkins.dynamicdevices.co.uk:8080/job/yocto/