DynamicDevices/meta-mono #58: Mono image tests fail

ajlennon, 2021-01-26:


ajlennon, 2021-01-26:
Any thoughts @zboszor ? This seems to relate to the 6.12 support?

zboszor, 2021-01-26:
@ajlennon Can you try with setting the preferred version to 6.10 temporarily?

zboszor, 2021-01-26:
This is a build issue in libgdiplus. The presence of `GdipCreateFromXDrawable_linux` is protected by:
```
#if defined(HAVE_X11) && CAIRO_HAS_XLIB_SURFACE
...
#endif
```
The former is not defined, the latter is defined as `1` by the Cairo headers.

zboszor, 2021-01-26:
And the bug is in configure.ac in libgdiplus:
```
AC_ARG_WITH(x11, 
        [AC_HELP_STRING([--without-x11], [Disable X11 support])],
        [disable_x11=yes])
```
With this, both `--with-x11` and `--without-x11` disables X11 support. Workaround is coming right up.

zboszor, 2021-01-26:
PR #59 fixed the issue for me, just by rebuilding and installing libgdiplus0.

ajlennon, 2021-01-28:
So.... I've had a successful built on Jenkins finally (thud, x86, 6.12)

http://jenkins.dynamicdevices.co.uk:8080/job/yocto/ARCH=x86,BRANCH=thud,VERSION=6.12.0.107/32/console

However other builds are still failing. Going to look at why, then gradually add in the other Yocto releases

http://jenkins.dynamicdevices.co.uk:8080/job/yocto/

ajlennon, 2021-01-28:
OK! So the thud x86 builds are OK. The ARM builds are failing though for 6.10 and 6.12

http://jenkins.dynamicdevices.co.uk:8080/job/yocto/ARCH=arm,BRANCH=thud,VERSION=6.10.0.104/33/console

ajlennon, 2021-01-28:
ARM 6.8 is OK so it looks like the problems came in with 6.10

http://jenkins.dynamicdevices.co.uk:8080/job/yocto/

ajlennon, 2021-01-28:
Some good news. ARM Warrior builds 6.10 so perhaps thud is just too old

http://jenkins.dynamicdevices.co.uk:8080/job/yocto/ARCH=arm,BRANCH=warrior,VERSION=6.10.0.104/35/console

ajlennon, 2021-01-30:
Pretty nice. Will add in dunfell and gatesgarth.

[image]

I think we could do with replacing the core image with an image with recipes that people also care about, like msbuild.

Also it would be good to have more test cases for things like msbuld. Fancy doing a simple msbuild test case @zboszor ?

e.g. https://github.com/DynamicDevices/meta-mono/blob/master/lib/oeqa/runtime/cases/mono.py

ajlennon, 2021-01-31:
There's something basic stopping the sumo builds running - Python version? 

Any thoughts @zboszor ?

http://jenkins.dynamicdevices.co.uk:8080/job/yocto/

zboszor, 2021-02-01:
I use CentOS 7 as the buildhost and I have python3 (3.6.8) installed. /usr/bin/python is still symlinked to python2 (2.7.5).

ajlennon, 2021-02-01:
The build box is Ubuntu 18.04.5 with python2 default and python3 installed

```
root@jenkins:~# python
Python 2.7.17 (default, Sep 30 2020, 13:38:04) 
[GCC 7.5.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> 
root@jenkins:~# python3
Python 3.6.9 (default, Oct  8 2020, 12:12:24) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```

ajlennon, 2021-02-04:
OK I am pretty happy with the build matrix now so closing this issue

[image]

closed 2021-02-04 by ajlennon