DynamicDevices/meta-mono #64: msbuild fails to build

ajlennon, 2021-02-04:


ajlennon, 2021-02-04:
Thoughts  @mnme @zboszor ?

mnme, 2021-02-04:
The dotnet recipe doesn't seem to be compatible with the i586 architecture. Or at least it's not tested against that. You can try to change the offending line in `dotnet_3.1.1.bb` and see if it builds.

Line 9 from `COMPATIBLE_HOST ?= "(x86_64).*-linux"` to `COMPATIBLE_HOST ?= "(i586|x86_64).*-linux"`.

If that doesn't help, there's also a new dotnet release which is compatible with more architectures [here](https://github.com/intel-iot-devkit/meta-iot-cloud/blob/master/recipes-devtools/dotnet/dotnet_5.0.101.bb). This one doesn't mention i586 either, but who knows.

ajlennon, 2021-02-04:
Hmm. Older Yocto fails with that. Newer Yocto fails in a different way...

```
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: Architecture did not match (x86-64, expected x86) on /work/i586-poky-linux/dotnet/3.1.11-r0/packages-split/dotnet-dev/usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost [arch]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: /usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost contained in package dotnet-dev requires libgcc_s.so.1(GCC_3.0)(64bit), but no providers found in RDEPENDS_dotnet-dev? [file-rdeps]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: /usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost contained in package dotnet-dev requires libdl.so.2(GLIBC_2.2.5)(64bit), but no providers found in RDEPENDS_dotnet-dev? [file-rdeps]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: /usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost contained in package dotnet-dev requires ld-linux-x86-64.so.2(GLIBC_2.3)(64bit), but no providers found in RDEPENDS_dotnet-dev? [file-rdeps]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: /usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost contained in package dotnet-dev requires libpthread.so.0(GLIBC_2.2.5)(64bit), but no providers found in RDEPENDS_dotnet-dev? [file-rdeps]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: /usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost contained in package dotnet-dev requires libstdc++.so.6(GLIBCXX_3.4.9)(64bit), but no providers found in RDEPENDS_dotnet-dev? [file-rdeps]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: /usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost contained in package dotnet-dev requires libstdc++.so.6(GLIBCXX_3.4.15)(64bit), but no providers found in RDEPENDS_dotnet-dev? [file-rdeps]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: /usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost contained in package dotnet-dev requires libstdc++.so.6(CXXABI_1.3)(64bit), but no providers found in RDEPENDS_dotnet-dev? [file-rdeps]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: /usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost contained in package dotnet-dev requires libstdc++.so.6(GLIBCXX_3.4.11)(64bit), but no providers found in RDEPENDS_dotnet-dev? [file-rdeps]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: /usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost contained in package dotnet-dev requires libstdc++.so.6(GLIBCXX_3.4)(64bit), but no providers found in RDEPENDS_dotnet-dev? [file-rdeps]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: /usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost contained in package dotnet-dev requires libc.so.6(GLIBC_2.14)(64bit), but no providers found in RDEPENDS_dotnet-dev? [file-rdeps]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: /usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost contained in package dotnet-dev requires libc.so.6(GLIBC_2.4)(64bit), but no providers found in RDEPENDS_dotnet-dev? [file-rdeps]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: /usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost contained in package dotnet-dev requires libc.so.6(GLIBC_2.3)(64bit), but no providers found in RDEPENDS_dotnet-dev? [file-rdeps]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: /usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost contained in package dotnet-dev requires libc.so.6(GLIBC_2.2.5)(64bit), but no providers found in RDEPENDS_dotnet-dev? [file-rdeps]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: /usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost contained in package dotnet-dev requires libpthread.so.0()(64bit), but no providers found in RDEPENDS_dotnet-dev? [file-rdeps]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: /usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost contained in package dotnet-dev requires libdl.so.2()(64bit), but no providers found in RDEPENDS_dotnet-dev? [file-rdeps]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: /usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost contained in package dotnet-dev requires libstdc++.so.6()(64bit), but no providers found in RDEPENDS_dotnet-dev? [file-rdeps]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: /usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost contained in package dotnet-dev requires libm.so.6()(64bit), but no providers found in RDEPENDS_dotnet-dev? [file-rdeps]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: /usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost contained in package dotnet-dev requires libgcc_s.so.1()(64bit), but no providers found in RDEPENDS_dotnet-dev? [file-rdeps]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: /usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost contained in package dotnet-dev requires libc.so.6()(64bit), but no providers found in RDEPENDS_dotnet-dev? [file-rdeps]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: /usr/share/dotnet/sdk/3.1.111/AppHostTemplate/apphost contained in package dotnet-dev requires ld-linux-x86-64.so.2()(64bit), but no providers found in RDEPENDS_dotnet-dev? [file-rdeps]
10:35:07 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: Architecture did not match (x86-64, expected x86) on /work/i586-poky-linux/dotnet/3.1.11-r0/packages-split/dotnet-dbg/usr/share/dotnet/shared/Microsoft.NETCore.App/3.1.11/.debug/createdump
[... cut ...]
10:35:07 Architecture did not match (x86-64, expected x86) on /work/i586-poky-linux/dotnet/3.1.11-r0/packages-split/dotnet-dbg/usr/share/dotnet/shared/Microsoft.NETCore.App/3.1.11/.debug/System.Globalization.Native.so [arch]
10:35:08 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: non -dev/-dbg/nativesdk- package contains symlink .so: dotnet path '/work/i586-poky-linux/dotnet/3.1.11-r0/packages-split/dotnet/usr/lib/libhostfxr.so' [dev-so]
10:35:08 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: Architecture did not match (x86-64, expected x86) on /work/i586-poky-linux/dotnet/3.1.11-r0/packages-split/dotnet/usr/share/dotnet/dotnet
```

ajlennon, 2021-02-04:
Where does this come from? Can we find an x86 equivalent?

```
SRC_URI = "https://download.visualstudio.microsoft.com/download/pr/d5d940c0-4c2f-4cbb-8d12-33bfb30c4db8/619b5be6e995cead6e9432134f903711/${BPN}-sdk-${SDK}-linux-x64.tar.>
```

mnme, 2021-02-04:
This link comes from the official [Microsoft dotnet core download page](https://dotnet.microsoft.com/download/dotnet-core/3.1). From a quick look it seems that they only provide an x86 binary for Windows…

ajlennon, 2021-02-04:
Oh dear - OK so can you remind me why we need the dotnet stuff for the msbuild recipe?

mnme, 2021-02-04:
I'm not sure which part exactly needs dotnet, as I never got a fond understanding of all the moving parts in the msbuild build process. I'm afraid that the whole thing is kind of a blackbox to me and I just patched everything that failed or was missing until it worked.

Inspiration was taken from the [gentoo msbuild 15.3 ebuild](https://github.com/gentoo/dotnet/blob/master/dev-util/msbuild/msbuild-15.3-r3.ebuild) and the Arch Linux [msbuild PKGBULD](https://github.com/archlinux/svntogit-community/blob/packages/mono-msbuild/trunk/PKGBUILD).

ajlennon, 2021-02-05:
Have moved the CI build to x86_64 and now I see this

```
04:16:53 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: non -dev/-dbg/nativesdk- package contains symlink .so: dotnet path '/work/core2-64-poky-linux/dotnet/3.1.11-r0/packages-split/dotnet/usr/lib/libhostfxr.so' [dev-so]
```

http://jenkins.dynamicdevices.co.uk:8080/job/yocto/ARCH=x86-64,BRANCH=sumo,VERSION=6.10.0.104/67/console

mnme, 2021-02-05:
I've never seen that QA error before, but I've only built it on zeus and dunfell so far. Maybe @zboszor knows more if this is somehow sumo specific?

ajlennon, 2021-02-05:
ok I have a zeus build here it did which seems the same

```
04:49:28 ERROR: dotnet-3.1.11-r0 do_package_qa: QA Issue: non -dev/-dbg/nativesdk- package contains symlink .so: dotnet path '/work/core2-64-poky-linux/dotnet/3.1.11-r0/packages-split/dotnet/usr/lib/libhostfxr.so' [dev-so]
```

http://jenkins.dynamicdevices.co.uk:8080/job/yocto/ARCH=x86-64,BRANCH=zeus,VERSION=6.12.0.107/67/console

ajlennon, 2021-02-05:
(btw. I am not trying to put hurdles in the way here. Just think it'll benefit us all if we can get a working CI test matrix running on PRs for the future)

mnme, 2021-02-06:
Fun fact: I can now reproduce the error. I'll send a pull request as soon as I have a good solution.

ajlennon, 2021-02-06:
Ah great! Thanks

PR #65 (2021-02-09): msbuild: Reduce dotnet dependency to libhostfxr
Fixes compatibility with x86 build hosts by building the library from
source and removes the dependency on the full dotnet package.

I would like to thank openSUSE for their excellent [msbuild-libhostfxr](https://build.opensuse.org/package/view_file/Mono:Factory/msbuild-libhostfxr/msbuild-libhostfxr.spec) package that put me on the right track.

Building for x86 is currently untested, if somebody has an x86 machine to test with, please do so and report back.

Fixes #64
closed 2021-02-09 by ajlennon