DynamicDevices/meta-mono #82: mono project changed branch name from master to main

iceaway, 2021-03-16:
Hi,

The mono project has changed the [master branch name to main](https://github.com/mono/mono/commit/83cbc378af35346ae538d787d46dac209e8fd6da), causing the mono_git.bb recipe to fail fetching the sources.

ajlennon, 2021-03-16:
Oh great. OK we'll have to update

iceaway, 2021-03-16:
Unfortunately I do not have a proper build environment set up for testing, otherwise I'd prepare a pull request.

ajlennon, 2021-03-16:
Appreciate the thought - I've added a git/master build to the Jenkins server so will progress this and revert :)

ajlennon, 2021-03-16:
Hmm - I think I removed mono git build from meta-mono master a while ago. What branch are you building?

iceaway, 2021-03-17:
We are on the 'sumo' branch currently. I know it's really old (and we are in progress of upgrading) but in the meantime it would be fantastic if you'd be willing to fix it from on the sumo branch.

ajlennon, 2021-03-17:
So I put a patch in there to use main but it seems unhappy trying to patch. When's the last time you built?

http://jenkins.dynamicdevices.co.uk:8080/job/yocto/ARCH=x86-64,BRANCH=sumo,VERSION=git/101/console

iceaway, 2021-03-17:
It just finished building successfully here with your latest commit on the sumo branch. But I'm not sure it actually rebuilt mono on our build server, it may be in our sstate cache already. I do not have deep enough knowledge about bitbake to know exactly what happens under the hood.

iceaway, 2021-03-17:
I cleaned the sstate-cache and rebuild, getting these warnings:
```WARNING: mono-native-5.12.0.226-r0 do_patch: 
Some of the context lines in patches were ignored. This can lead to incorrectly applied patches.
The context lines in the patches can be updated with devtool:

    devtool modify <recipe>
    devtool finish --force-patch-refresh <recipe> <layer_path>

Then the updated patches and the source tree (in devtool's workspace)
should be reviewed to make sure the patches apply in the correct place
and don't introduce duplicate lines (which can, and does happen
when some of the context is ignored). Further information:
http://lists.openembedded.org/pipermail/openembedded-core/2018-March/148675.html
https://bugzilla.yoctoproject.org/show_bug.cgi?id=10450
Details:
Applying patch dllmap-config.in.diff
patching file data/config.in
Hunk #1 succeeded at 10 with fuzz 2.
Hunk #2 succeeded at 36 (offset 6 lines).

Now at patch dllmap-config.in.diff
NOTE: recipe mono-native-5.12.0.226-r0: task do_patch: Succeeded
NOTE: Running task 6573 of 6602 (/var/jenkins_home/workspace/Kodkod_master/sources/meta-mono/recipes-mono/mono/mono-native_5.12.0.226.bb:do_configure)
NOTE: Running task 6574 of 6602 (/var/jenkins_home/workspace/Kodkod_master/sources/meta-mono/recipes-mono/mono/mono-native_5.12.0.226.bb:do_populate_lic)
NOTE: recipe mono-native-5.12.0.226-r0: task do_configure: Started
NOTE: recipe mono-native-5.12.0.226-r0: task do_populate_lic: Started
NOTE: recipe mono-native-5.12.0.226-r0: task do_populate_lic: Succeeded
NOTE: recipe mono-5.12.0.226-r0: task do_unpack: Succeeded
NOTE: Running task 6575 of 6602 (/var/jenkins_home/workspace/Kodkod_master/sources/meta-mono/recipes-mono/mono/mono_5.12.0.226.bb:do_patch)
NOTE: recipe mono-5.12.0.226-r0: task do_patch: Started
WARNING: mono-5.12.0.226-r0 do_patch: 
Some of the context lines in patches were ignored. This can lead to incorrectly applied patches.
The context lines in the patches can be updated with devtool:

    devtool modify <recipe>
    devtool finish --force-patch-refresh <recipe> <layer_path>

Then the updated patches and the source tree (in devtool's workspace)
should be reviewed to make sure the patches apply in the correct place
and don't introduce duplicate lines (which can, and does happen
when some of the context is ignored). Further information:
http://lists.openembedded.org/pipermail/openembedded-core/2018-March/148675.html
https://bugzilla.yoctoproject.org/show_bug.cgi?id=10450
Details:
Applying patch dllmap-config.in.diff
patching file data/config.in
Hunk #1 succeeded at 10 with fuzz 2.
Hunk #2 succeeded at 36 (offset 6 lines).
```

Obviously we are not building the git version of the recipe, not sure why, I do not think we specify force a specific version of mono anywhere.

ajlennon, 2021-03-17:
Unless you change it it will default to the layer default which is here for `sumo`

https://github.com/DynamicDevices/meta-mono/blob/a230762cc6480a774da0bdad9d98f594bce5a32e/conf/layer.conf#L13

iceaway, 2021-03-18:
I see, then I understand why it picks up that version. I'm not sure there is much else I can do unfortunately, I am not familiar with the mono code base at all, I only install it as a package on our file system as a dependency.

ajlennon, 2021-03-18:
Hiya - I'm a bit unclear about what's building and what's broken? What do you actually need?

iceaway, 2021-03-18:
I am a bit confused myself. Before your newest commit, our build would fail with this message:
```
ERROR: ExpansionError during parsing /var/jenkins_home/workspace/Kodkod_master/sources/meta-mono/recipes-mono/mono/mono_git.bb
Traceback (most recent call last):
bb.data_smart.ExpansionError: Failure expanding variable SRCPV, expression was ${@bb.fetch2.get_srcrev(d)} which triggered exception FetchError: Fetcher failure: Unable to resolve 'master' in upstream git repository in git ls-remote output for github.com/mono/mono.git
```
But after your changes, this error message went away. Since we do not build from latest git, but rather the default version for the layer which you specified above, we do not get the patch errors. I'm not sure why this is happening, maybe it looks at the git recipe even though it does not use it?

ajlennon, 2021-03-18:
Very very strange. Maybe it does!

So with the change to mono_git so it references the new branch are you building ok?

iceaway, 2021-03-18:
Yes, everything is working fine here so I'm as happy as I can be!

ajlennon, 2021-03-18:
Great !!!

closed 2021-03-18 by ajlennon