DynamicDevices/meta-mono #85: Secure Socket Layer (SSL) / Transport Layer Security (TLS)

befedo, 2021-04-21:
Hi there,

Like described [here](https://www.mono-project.com/docs/faq/security/) I have problems running an application which calls `System.Net.WebClient.DownloadString`. The mono runtime yields an exception like:
```{text}
Mono.Btls.MonoBtlsException: Ssl error:1000007d:SSL routines:OPENSSL_internal:CERTIFICATE_VERIFY_FAILED
   at /usr/src/debug/mono/6.12.0.107-r0/mono-6.12.0.107/external/boringssl/ssl/handshake_client.c:1132
   at Mono.Btls.MonoBtlsContext.ProcessHandshake () [0x00048] in <94d2e1fff46b45c0a06833385db5f572>:0 
   at Mono.Net.Security.MobileAuthenticatedStream.ProcessHandshake (Mono.Net.Security.AsyncOperationStatus status, System.Boolean renegotiate) [0x000da] in <94d2e1fff46b45c0a06833385db5f572>:0 
   at (wrapper remoting-invoke-with-check) Mono.Net.Security.MobileAuthenticatedStream.ProcessHandshake(Mono.Net.Security.AsyncOperationStatus,bool)
   at Mono.Net.Security.AsyncHandshakeRequest.Run (Mono.Net.Security.AsyncOperationStatus status) [0x00006] in <94d2e1fff46b45c0a06833385db5f572>:0 
   at Mono.Net.Security.AsyncProtocolRequest.ProcessOperation (System.Threading.CancellationToken cancellationToken) [0x000fc] in <94d2e1fff46b45c0a06833385db5f572>:0
```
The certificate used ist installed via `ca-certificates` and
```{bash}
# curl --insecure -vvI https://www.google.com/
```
yields `* Connection #0 to host www.google.com left intact` (google.com as example) so the certificates where used by the system.

So the question arises, how doe we install/deploy the mono trust under `/usr/share/.mono/certs`?
On the Target machine I don't have `cert-sync`, `certmgr.exe` available (or couldn't find it).

Thanks for your time and best regards
befedo

befedo, 2021-04-21:
In the meantime I've found `mozroot-certdata` which installed all certificates to `/usr/share/.mono/certs` but the exception persists.

befedo, 2021-04-21:
Okay, I think I figured it out ...

`mozroot-certdata` is pretty outdated, but I finally found `cert-sync` and running:
```{text}
root@intel-corei7-64:/usr/lib/mono/4.5# mono cert-sync.exe /etc/ssl/certs/ca-certificates.crt 
Mono Certificate Store Sync - version 6.12.0.107
Populate Mono certificate store from a concatenated list of certificates.
Copyright 2002, 2003 Motus Technologies. Copyright 2004-2008 Novell. BSD licensed.

Importing into legacy system store:
I already trust 251, your new list has 129
Certificate added: C=FR, O=Certinomis, OU=0002 433998903, CN=Certinomis - Root CA
Certificate added: C=FR, O=Certplus, CN=Class 2 Primary CA
Certificate added: C=RO, O=certSIGN, OU=certSIGN ROOT CA
...
```
Imported all certificates from the `ca-certificates` package on the target machine.
Now I need to figure out how to do this on a recipe basis.

Sorry for the noise,
befedo

closed 2021-04-21 by befedo

ajlennon, 2021-04-21:
Not at all - thanks for the info! PR's on this gratefully accepted @befedo !

ajlennon, 2021-04-21:
Also I am not sure if this is appropriate but reaching back into the mists of time there is a mechanism by which you can tell Yocto to run scripting on the target on first boot for setup. That _might_ be a way of doing this?

befedo, 2021-04-21:
@ajlennon For now I picked from the `mozroot-certdata` something like this:
```{bash}
pkg_postinst_${PN} () {
     #!/bin/sh -e
     if [ x"$D" = "x" ]; then
          mono /usr/lib/mono/4.5/cert-sync.exe --quiet ${sysconfdir}/ssl/certs/ca-certificates.crt
     else
          exit 1
     fi
}
```
but need to test it on the target machine on fresh install/update of package.

ajlennon, 2021-04-21:
Yeah nice :)