DynamicDevices/meta-mono #97: add dotnet 5 recipe for dunfell branch

karthago1, 2021-10-15:
this is my first merge request, so I hope I did everything in the correct way.
this recipe is tested under dunfell branch and is copied from http://github.com/intel-iot-devkit/meta-iot-cloud/blob/master/recipes-devtools/dotnet/dotnet_5.0.300.bb

it will be nice if it will be included to the meta-mono (dunfell branch)

ajlennon, 2021-10-15:
I'd like us to get this into the master branch if possible. Have you tried building against that?

CTerasaGuF, 2022-02-23:
Not sure how this can be marked as "tested" at least for dunfell. The recipe obviously uses the more recent honister override syntax. See https://docs.yoctoproject.org/migration-guides/migration-3.4.html#override-syntax-changes
A plain old dunfell build does not work for me. But it should not be too hard to fix this.

ajlennon, 2022-02-23:
These are the latest (building and tested) dunfell builds

https://github.com/DynamicDevices/meta-mono/actions/runs/1794025332

CTerasaGuF, 2022-02-23:
Yes, I get the following error, but perhaps I am not using the latest poky dunfell version:
```ERROR: ParseError at /home/clemens.terasa/dunfell/sources/poky/meta-mono/recipes-mono/dotnet/dotnet_5.0.300.bb:15: unparsed line: 'RDEPENDS:${PN} = "    icu     libgssapi-krb5     lttng-ust     zlib "'```

I am preparing a patch for this that works for me and will propose a MR.

karthago1, 2022-02-23:
yes you are right. I have no Idea why it is working with my dunfell branch. I will try to find out the reason