DynamicDevices/meta-mono #103: Add mono 6.8 and msbuild 16.6 to morty

klasyc, 2021-12-02:
Hi,

I am using EOL branch `morty` and I needed to update some recipes, namely `msbuild` to version 16.6 and `mono` to version 6.8. This operation also involves some dependencies. Now both recipes compile correctly and I would like to share my changes to the community. Please see my PR for details. This is my very first Yocto contribution, please be lenient with me...

Klasyc

ajlennon, 2022-01-29:
I have the autobuilder going now. The build is failing. Need to work out why. It's probably related to the host build environment being too new. The older builds really don't seem to like gcc 9/10

klasyc, 2022-01-29:
I am using Ubuntu 16.04 Docker image for my builds. Thank you for paying attention to my work :)

ajlennon, 2022-02-07:
Merged

closed 2022-02-07 by ajlennon