DynamicDevices/meta-mono #105: Yocto Zeus and Gatesgarth issues with Mono 6.10 and 6.12  Large cursor on TextBox

lbarrosoneto, 2021-12-22:
## Steps to Reproduce

1.   Compile Mono 6.10.0.104 from source (as part of the  Yocto Linux image build) using the Meta Layer https://git.yoctoproject.org/meta-mono for gatesgarth branch 
2.  Execute the TxtBoxTestApp.exe by typing: DISPLAY=:0.0 mono TxtBoxTestApp.exe
3. Issue: Cursor is way bigger than fonts, making the text box lines way higher than necessary.

### Current Behavior

TextBox is display a cursor way bigger than the font (this is happening on Windriver Linux)
[image]

Issue does not happen on Debian Linux.

### Expected Behavior

Cursor should be same size of the fonts used in TextBox (this is on a Windows Host):
[image]

## On which platforms did you notice this

[ ] macOS
[ x] Linux - Yocto Zeus or Gatesgarth
[ ] Windows

**Version Used**:

6.10.0.104 

## Stacktrace

Not applicable.

## Remark

The issue does not happen with Mono Version 6.8.0.123.

lbarrosoneto, 2021-12-22:
[TxtBoxTestApp.exe.zip](https://github.com/DynamicDevices/meta-mono/files/7764109/TxtBoxTestApp.exe.zip)

Attaching test application

lbarrosoneto, 2021-12-22:
As reported  by @otavio on https://github.com/mono/mono/issues/21367, this is yocto specific and does not happen on Debian or NixOS

mnme, 2022-01-17:
I've seen something similar in the past. If I remember correctly, downgrading libgdiplus to version 6.0.2 solved it for me.

otavio, 2022-01-17:
@mnme what is interesting is it works on Desktop so something more subtle is impacting it.

ajlennon, 2022-01-29:
Is master still doing this?

lbarrosoneto, 2022-02-01:
@ajlennon 
I am currently having issues building the mono layer, from master, using a ubuntu20.04 LTS machine.
Yocto is using gatesgarth, Mono from Master.

Build Configuration:
BB_VERSION           = "1.48.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "x86_64-poky-linux"
MACHINE              = "intel-corei7-64"
DISTRO               = "poky"
DISTRO_VERSION       = "3.2.4"
TUNE_FEATURES        = "m64 corei7"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "gatesgarth:bc71ec0f1d97c983cb0a089b15e283c56de30844"
meta-intel           = "gatesgarth:adf3288b9890cbeb2519b57ca548a025ef7a82b0"
meta-virtualization  = "gatesgarth:2e5df1cd76265587e6db67f020a06566c4ed6dc4"
meta-oe              
meta-networking      
meta-python          
meta-perl            
meta-filesystems     
meta-xfce            
meta-gnome           
meta-multimedia      = "gatesgarth:f3f7a5f1a4713f145107bb043e0d14cb3a51c62f"
meta-tpm             = "gatesgarth:3daf99fd138b0eebe864bbe1b9c71241d97c4512"
meta-mono            = "master:c93841d4c5afc9808840cd4c0ae0697c6c3ef7e0"

make[4]: Entering directory '/home1/luiz/gatesgarth/yocto/tmp/work/x86_64-linux/mono-native/6.12.0.161-r0/mono-6.12.0.122/mono/eglib'
../../doltlibtool  --tag=CC   --mode=link gcc  -g -Wall -D_GNU_SOURCE -D_FORTIFY_SOURCE=2 -Wall -Wunused -Wmissing-declarations -Wpointer-arith -Wno-cast-qual -Wwrite-strings -Wno-switch -Wno-switch-enum -Wno-unused-value -Wno-attributes -isystem/home1/luiz/gatesgarth/yocto/tmp/work/x86_64-linux/mono-native/6.12.0.161-r0/recipe-sysroot-native/usr/include -O2 -pipe -Wmissing-prototypes -Wstrict-prototypes -Wnested-externs -Wno-format-zero-length -std=gnu99 -fno-strict-aliasing -fwrapv -DMONO_DLL_EXPORT -g -mno-tls-direct-seg-refs -Werror=return-type -Werror-implicit-function-declaration   -L/home1/luiz/gatesgarth/yocto/tmp/work/x86_64-linux/mono-native/6.12.0.161-r0/recipe-sysroot-native/usr/lib                         -L/home1/luiz/gatesgarth/yocto/tmp/work/x86_64-linux/mono-native/6.12.0.161-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/home1/luiz/gatesgarth/yocto/tmp/work/x86_64-linux/mono-native/6.12.0.161-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/home1/luiz/gatesgarth/yocto/tmp/work/x86_64-linux/mono-native/6.12.0.161-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/home1/luiz/gatesgarth/yocto/tmp/work/x86_64-linux/mono-native/6.12.0.161-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/home1/luiz/gatesgarth/yocto/tmp/work/x86_64-linux/mono-native/6.12.0.161-r0/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home1/luiz/gatesgarth/yocto/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 -Wl,--export-dynamic -Wl,-Bsymbolic -Wl,-z,now -o libeglib.la  libeglib_la-garray.lo libeglib_la-gbytearray.lo libeglib_la-gerror.lo libeglib_la-ghashtable.lo libeglib_la-giconv.lo libeglib_la-gmem.lo libeglib_la-goutput.lo libeglib_la-gqsort.lo libeglib_la-gstr.lo libeglib_la-gslist.lo libeglib_la-gstring.lo libeglib_la-gptrarray.lo libeglib_la-glist.lo libeglib_la-gqueue.lo libeglib_la-gpath.lo libeglib_la-gshell.lo libeglib_la-gspawn.lo libeglib_la-gfile.lo libeglib_la-gfile-posix.lo libeglib_la-gpattern.lo libeglib_la-gmarkup.lo libeglib_la-gutf8.lo libeglib_la-gunicode.lo libeglib_la-gdate-unix.lo libeglib_la-gdir-unix.lo libeglib_la-gfile-unix.lo libeglib_la-gmisc-unix.lo libeglib_la-gmodule-unix.lo libeglib_la-gtimer-unix.lo libeglib_la-gmodule-aix.lo -lm  -lm -lm -lrt  -ldl -lpthread
../../doltlibtool: line 19: /home1/luiz/gatesgarth/yocto/tmp/work/x86_64-linux/mono-native/6.12.0.161-r0/mono-6.12.0.122/mono/eglib/../../libtool: No such file or directory
make[4]: *** [Makefile:720: libeglib.la] Error 127
make[4]: Leaving directory '/home1/luiz/gatesgarth/yocto/tmp/work/x86_64-linux/mono-native/6.12.0.161-r0/mono-6.12.0.122/mono/eglib'
make[3]: *** [Makefile:1089: all-recursive] Error 1
make[3]: Leaving directory '/home1/luiz/gatesgarth/yocto/tmp/work/x86_64-linux/mono-native/6.12.0.161-r0/mono-6.12.0.122/mono/eglib'

ajlennon, 2022-02-01:
Thanks - I'm doing a lot of work at the moment to try to make all layers pass tests which I believe will be useful for the future.

Unfortunately this involves change and perhaps some pain for now.

To avoid this - pin your use of meta-mono to what you have been using and none of this will affect you.

Please look at the read me for what is building and testing successfully as this will be the source of truth moving forward. 

We have a system now for building with standard host environments which are here - 

https://github.com/DynamicDevices/yocto-ci-build

So the idea moving forward is master will always build and test against a standard host build environment for MACHINE={qemux86,qemuarm} 

I would love to see contributions on ideas for how to improve build and test for us all

lbarrosoneto, 2022-02-02:
@ajlennon 
Looks like build does not fail from this tag: https://github.com/DynamicDevices/meta-mono/commit/da01212939e79ef0d2e5fe81c55d764a4f8f7205

Loading cache: 100% |                                                                            | ETA:  --:--:--
Loaded 0 entries from dependency cache.
Parsing recipes: 100% |###########################################################################| Time: 0:00:24
Parsing of 2916 .bb files complete (0 cached, 2916 parsed). 4237 targets, 130 skipped, 0 masked, 0 errors.
Removing 1 recipes from the corei7-64 sysroot: 100% |#############################################| Time: 0:00:03
Removing 1 recipes from the x86_64 sysroot: 100% |################################################| Time: 0:00:03
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.48.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "x86_64-poky-linux"
MACHINE              = "intel-corei7-64"
DISTRO               = "poky"
DISTRO_VERSION       = "3.2.4"
TUNE_FEATURES        = "m64 corei7"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "gatesgarth:bc71ec0f1d97c983cb0a089b15e283c56de30844"
meta-intel           = "gatesgarth:adf3288b9890cbeb2519b57ca548a025ef7a82b0"
meta-virtualization  = "gatesgarth:2e5df1cd76265587e6db67f020a06566c4ed6dc4"
meta-oe              
meta-networking      
meta-python          
meta-perl            
meta-filesystems     
meta-xfce            
meta-gnome           
meta-multimedia      = "gatesgarth:f3f7a5f1a4713f145107bb043e0d14cb3a51c62f"
meta-tpm             = "gatesgarth:3daf99fd138b0eebe864bbe1b9c71241d97c4512"
meta-mono            = "HEAD:da01212939e79ef0d2e5fe81c55d764a4f8f7205"

Given the nature of the error I am experiencing when using yocto gatesgarth + meta-mono from master, I suspect this might be a breaking change when using gatesgarth.

https://github.com/DynamicDevices/meta-mono/commit/588fe511a32cdf03a414e3d1fb2ff7eb91e9c463
[image]

ajlennon, 2022-02-02:
So, if you look at the README gatesgarth is building successfully. 

This is the build run - https://github.com/DynamicDevices/meta-mono/actions/runs/1777249145

This is the container image that builds it - https://github.com/DynamicDevices/meta-mono/blob/master/.github/workflows/CI_github.yml#L14

And what is in that image is defined here - https://github.com/DynamicDevices/yocto-ci-build

So from my end this is "working" atm.

lbarrosoneto, 2022-02-02:
@ajlennon 

Sorry, I think I was doing the build a non-traditional way.
Yocto gatesgarth with meta-mono from gatesgarth branch: Builds ok
Yocto gatesgarth with meta-mono from master: Builds if I take out the commit I mentioned above.

A few things I will try and report back:
- Check if the 2 builds still experience the TextBox issue.
- Try a build from Yocto master + meta-mono master and check if Textbox issue still happens.

lbarrosoneto, 2022-02-03:
@ajlennon 

I just confirmed that mono issue with TextBoxes is still happening on Master

Build Configuration:
BB_VERSION           = "1.53.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "x86_64-poky-linux"
MACHINE              = "qemux86-64"
DISTRO               = "poky"
DISTRO_VERSION       = "3.4+snapshot-74a8a74a553a33dc5f41939f8070d75e6d57d3da"
TUNE_FEATURES        = "m64 core2"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "master:74a8a74a553a33dc5f41939f8070d75e6d57d3da"
meta-intel           = "master:76cc0cce2c8cbe547009573901a1d7247e43fc97"
meta-oe              
meta-networking      
meta-python          
meta-perl            
meta-filesystems     
meta-xfce            
meta-gnome           
meta-multimedia      = "master:c597a154dd66c615aa636b071bb8220904ee19a5"
meta-tpm             = "master:deeadddfb4539d9fe3b78ef704e279ac4fc1e07b"
meta-mono            = "master:c93841d4c5afc9808840cd4c0ae0697c6c3ef7e0"

[image]

lbarrosoneto, 2022-02-09:
@mnme 
@otavio 
@ajlennon 

We did some additional experimentation while chasing another bug.

Tested on a x86_64 for on zeus branch:
Copying the libgdi plus binary to yocto seems to solve the issue when using mono 6.10.
This was the binary used: libgdiplus_6.0.5-0xamarin1+debian10b1.debian.tar.xz
Sources: https://download.mono-project.com/repo/debian/dists/stable-buster/snapshots/6.8.0.123/main/source/
File is attached.
[libgdiplus.so.0.0.0.zip](https://github.com/DynamicDevices/meta-mono/files/8028384/libgdiplus.so.0.0.0.zip)

Will do some testing on ARM.

So issue seems to be localized to libgdi, as @mnme  suspected.

ajlennon, 2022-02-09:
Interesting @lbarrosoneto and thanks for the update. I've been working hard to get a CI self-hosted runner system up and running which is pretty much done now.

So I guess my question is - is this still an issue with hardknott and master ?

ajlennon, 2022-02-09:
I am also very interested to know if you can think of an automated test for this we can add in

What do you get if you print out the size properties for those incorrect boxes?

svenluther, 2022-02-09:
Hi ...

We found out that pango was responsible for this, which is used in the layer for compiling libgdiplus. By disabling pango, the cursor is again normal size. 

In fact this seems logical, since pango is then used to measure fonts (and draw them) instead of cairo, and it seems there is an issue in the way pango measures the font for the (empty char ?) cursor.

Anyway, the following bbappend fixed it:

$ more libgdiplus_6.0.5.bbappend
SRC_URI_remove = "file://0001-configure-Fix-build-with-Gentoo-pango.patch"

PACKAGECONFIG = "${@bb.utils.contains('DISTRO_FEATURES', 'x11', 'x11', '', d)} jpeg tiff gif exif"
PACKAGECONFIG[pango] = "--with-pango,,pango"

ajlennon, 2022-02-09:
Excellent thanks. Can you provide some PRs into the relevant branches which will kick off the automated build & test cycle. When that passes I'll merge in

ajlennon, 2022-02-09:
(Bonus points if you can provide some C# source that'll do a simple test...)

ajlennon, 2022-02-09:
Hi there - thanks for the PRs. I'm not sure if I am misunderstanding but normally I would expect a single PR to include a range of commits, not have one commit per PR. Just makes it a lot easier to see what's going on and makes build failures less likely

lbarrosoneto, 2022-02-09:
@ajlennon 

Updating the app to show the text box height and will share here.

Luiz

ajlennon, 2022-02-09:
No worries at all. I appreciate the contribution. We’re all learning as we go. 

What I tend to do is

- fork the repo to my own space
- checkout the branch I want to work on
- create a new branch for my changes 
- make multiple fairly granular commits with blocks of changes
- try to make sure each commit is self contained and won’t break anything 
- Push my branch to the repo in my space
- Create a PR to the upstream repo (in this case this one) with some details of what it’s Ll about and ideally # tagged against an existing issue which I or somebody raised to work against) 
- That’ll trigger the CI bulld and test cycle 
-  Discuss / iterate changes to the existing PR
-  PR gets committed 

Cheers

ajlennon, 2022-02-09:
So your first PR failed to build. Could you go through the process above and combine your commits into a single PR and then we can check that it builds? Thanks!

ajlennon, 2022-02-10:
@svenluther - take a look here. It's saying there's a cairo include file missing. Maybe a dependency missing?

https://github.com/DynamicDevices/meta-mono/runs/5128432001?check_suite_focus=true#step:7:11749

svenluther, 2022-02-10:
yeah, i don't know why this happens, on my own build it pulls in cairo all by itself, but i guess it may be pulled in by something else maybe ? 

Cairo is used as default if --with-pango is not used, so the error makes sense.

ajlennon, 2022-02-10:
Looks to me as though you need a dependency on cairo-dev maybe?

[image]

svenluther, 2022-02-16:
my .bbappend does :+1: 
$ more libgdiplus_6.0.5.bbappend
SRC_URI_remove = "file://0001-configure-Fix-build-with-Gentoo-pango.patch"

PACKAGECONFIG = "${@bb.utils.contains('DISTRO_FEATURES', 'x11', 'x11', '', d)} jpeg tiff gif exif"
PACKAGECONFIG[pango] = "--with-pango,,pango"

We need to remove the patch too which is in the libgdiplus_6.0.5.bb (and related) files.

How can i extend the PR to include this ?

svenluther, 2022-02-16:
no you need to drop the patch.

ajlennon, 2022-02-17:
I've closed your older PRs and your latest PR is building...

ajlennon, 2022-02-17:
Great stuff. That build and test worked - merged in

ajlennon, 2022-02-17:
Do you need to submit any PRs for other branches?

svenluther, 2022-02-18:
> [Please note: This e-mail is from an EXTERNAL e-mail address]
>
> Shall we close this issue or do you want to submit more PRs @svenluther<https://urldefense.com/v3/__https://github.com/svenluther__;!!AjveYdw8EvQ!PIorJcgZywBQZ3YCTWyj100jc7AnVuAovAIFhvWPCcei3BQvoHEP_xQoQxYEan71Wks$> ?

I think there where three branches affected on top of master.

I didn’t find the way to easily duplicate the PR for the other branches, so I need to find time to do this (unless you know of a magic way to do this, in which case you are welcome to handle it).

If not, I will  try to do the remaining PRs easy next week.

Friendly,

Sven Luther

ajlennon, 2023-04-03:
Closing due to inactivity