DynamicDevices/meta-mono #107: mono-xsp recipe build failing

avlec, 2022-01-20:
I am trying to include mono-xsp in a image for my target device. Mono itself builds fine and works on the hardware however, I haven't been able to get mono-xsp to build. I am unfamiliar with bitbake, and can't seem to find a solution in the layer files to this myself.

My local.conf was changed to this to support the packages.

`IMAGE_INSTALL_append = " mono mono-xsp ... "`

The error I am getting is the following, and I have installed libtool and libtool-bin on my development environment.

```
DEBUG: Executing shell function autotools_preconfigure
DEBUG: Shell function autotools_preconfigure finished
DEBUG: Executing python function autotools_aclocals
DEBUG: SITE files ['endian-little', 'bit-32', 'arm-common', 'arm-32', 'common-linux', 'common-glibc', 'arm-linux', 'arm-linux-gnueabi', 'common']
DEBUG: Python function autotools_aclocals finished
DEBUG: Executing shell function do_configure
Mono Certificate Store Sync - version 6.12.0.107
Populate Mono certificate store from a concatenated list of certificates.
Copyright 2002, 2003 Motus Technologies. Copyright 2004-2008 Novell. BSD licensed.

Importing into legacy user store:
I already trust 127, your new list has 127
Import process completed.

Importing into BTLS user store:
I already trust 127, your new list has 127
Import process completed.

**Error**: You must have `libtool' installed to compile mono_fpm.
Get ftp://ftp.gnu.org/pub/gnu/libtool-1.2d.tar.gz
(or a newer version if it is available)
Error: Exiting
Configure failed. The contents of all config.log files follows to aid debugging
ERROR: oe_runconf failed
WARNING: exit code 1 from a shell command.
ERROR: Execution of '.../tmp/work/armv7vet2hf-vfpv4d16-poky-linux-gnueabi/mono-xsp/git-r0/temp/run.do_configure.21388' failed with exit code 1
```

ajlennon, 2022-01-29:
I've updated the branch to `main` on master and am now getting this

[image]

ajlennon, 2023-04-03:
Closing due to inactivity

closed 2023-04-03 by ajlennon