DynamicDevices/meta-mono #111: master is failing to build - honister is ok - with 6.0.12.154

ajlennon, 2022-01-24:
I've seen this before but not sure why it's returned. Something to do with autotools / autotools-brokensep

```
| ../../doltlibtool  --tag=CC   --mode=link gcc  -g -Wall -D_GNU_SOURCE -D_FORTIFY_SOURCE=2 -Wall -Wunused -Wmissing-declarations -Wpointer-arith -Wno-cast-qual -Wwrite-strings -Wno-switch -Wno-switch-enum -Wno-unused-value -Wno-attributes -isystem/home/github/_work/meta-mono/meta-mono/master/build/tmp/work/x86_64-linux/mono-native/6.12.0.154-r0/recipe-sysroot-native/usr/include -O2 -pipe -Wmissing-prototypes -Wstrict-prototypes -Wnested-externs -Wno-format-zero-length -std=gnu99 -fno-strict-aliasing -fwrapv -DMONO_DLL_EXPORT -Wno-unused-but-set-variable -g -mno-tls-direct-seg-refs -Werror=incompatible-pointer-types -Werror=return-type -Werror-implicit-function-declaration   -L/home/github/_work/meta-mono/meta-mono/master/build/tmp/work/x86_64-linux/mono-native/6.12.0.154-r0/recipe-sysroot-native/usr/lib                         -L/home/github/_work/meta-mono/meta-mono/master/build/tmp/work/x86_64-linux/mono-native/6.12.0.154-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/home/github/_work/meta-mono/meta-mono/master/build/tmp/work/x86_64-linux/mono-native/6.12.0.154-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/home/github/_work/meta-mono/meta-mono/master/build/tmp/work/x86_64-linux/mono-native/6.12.0.154-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/home/github/_work/meta-mono/meta-mono/master/build/tmp/work/x86_64-linux/mono-native/6.12.0.154-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/home/github/_work/meta-mono/meta-mono/master/build/tmp/work/x86_64-linux/mono-native/6.12.0.154-r0/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/github/_work/meta-mono/meta-mono/master/build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 -Wl,--export-dynamic -Wl,-Bsymbolic -Wl,-z,now -o libeglib.la  libeglib_la-garray.lo libeglib_la-gbytearray.lo libeglib_la-gerror.lo libeglib_la-ghashtable.lo libeglib_la-giconv.lo libeglib_la-gmem.lo libeglib_la-goutput.lo libeglib_la-gqsort.lo libeglib_la-gstr.lo libeglib_la-gslist.lo libeglib_la-gstring.lo libeglib_la-gptrarray.lo libeglib_la-glist.lo libeglib_la-gqueue.lo libeglib_la-gpath.lo libeglib_la-gshell.lo libeglib_la-gspawn.lo libeglib_la-gfile.lo libeglib_la-gfile-posix.lo libeglib_la-gpattern.lo libeglib_la-gmarkup.lo libeglib_la-gutf8.lo libeglib_la-gunicode.lo libeglib_la-gdate-unix.lo libeglib_la-gdir-unix.lo libeglib_la-gfile-unix.lo libeglib_la-gmisc-unix.lo libeglib_la-gmodule-unix.lo libeglib_la-gtimer-unix.lo libeglib_la-gmodule-aix.lo -lm  -lm -lm -lrt  -ldl -lpthread
| ../../doltlibtool: line 19: /home/github/_work/meta-mono/meta-mono/master/build/tmp/work/x86_64-linux/mono-native/6.12.0.154-r0/mono-6.12.0.122/mono/eglib/../../x86_64-linux-libtool: No such file or directory
| make[4]: *** [Makefile:723: libeglib.la] Error 127
| make[4]: Leaving directory '/home/github/_work/meta-mono/meta-mono/master/build/tmp/work/x86_64-linux/mono-native/6.12.0.154-r0/mono-6.12.0.122/mono/eglib'
| make[3]: *** [Makefile:1092: all-recursive] Error 1
| make[3]: Leaving directory '/home/github/_work/meta-mono/meta-mono/master/build/tmp/work/x86_64-linux/mono-native/6.12.0.154-r0/mono-6.12.0.122/mono/eglib'
| make[2]: *** [Makefile:522: all-recursive] Error 1
| make[2]: Leaving directory '/home/github/_work/meta-mono/meta-mono/master/build/tmp/work/x86_64-linux/mono-native/6.12.0.154-r0/mono-6.12.0.122/mono'
| make[1]: *** [Makefile:592: all-recursive] Error 1
| make[1]: Leaving directory '/home/github/_work/meta-mono/meta-mono/master/build/tmp/work/x86_64-linux/mono-native/6.12.0.154-r0/mono-6.12.0.122'
| make: *** [Makefile:520: all] Error 2
| WARNING: exit code 2 from a shell command.
NOTE: recipe mono-native-6.12.0.154-r0: task do_compile: Failed
NOTE: recipe qemu-system-native-6.2.0-r0: task do_configure: Succeeded
NOTE: recipe gcc-11.2.0-r0: task do_compile: Succeeded
NOTE: Tasks Summary: Attempted 7239 tasks of which 6709 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/github/_work/meta-mono/meta-mono/master/meta-mono/recipes-mono/mono/mono-native_6.12.0.154.bb:do_compile
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
Error: Process completed with exit code 1.```

Any thoughts @zboszor ?

zboszor, 2022-01-24:
I have also seen this with my custom `phpNN` recipes, which are devised so e.g. PHP 7.3.x a.k.a. php73 and other versions can be installed in parallel.

I haven't had time to track it down yet.

ajlennon, 2022-01-24:
Good - thanks - not just me then!

zboszor, 2022-01-24:
It looks like Yocto 3.5 Kirkstone will not rename the libtool script anymore, so recipes need to adapt like this:
https://lists.openembedded.org/g/openembedded-devel/message/94322

ajlennon, 2022-01-24:
Brilliant - thanks for finding this. I've put in a patch and I think mono-native / mono is building now. It's now failing on msbuild it seems

closed 2022-01-24 by ajlennon