DynamicDevices/meta-mono #117: msbuild fails to build in dunfell and sumo

ajlennon, 2022-02-02:


snke, 2022-04-06:
Yes, same here:

```
| ** Using mono from /build/tmp/work/corei7-64-poky-linux/msbuild/15.4-r0/recipe-sysroot-native/usr/bin/mono                                                                                                                                 | + /build/tmp/work/corei7-64-poky-linux/msbuild/15.4-r0/recipe-sysroot-native/usr/bin/mono --version                                                                                                                                        | Mono JIT compiler version 6.10.0.104 (tarball Mon 04 Apr 2022 02:28:15 PM UTC)                                                                                                                                                             
| Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com                                                                                                                                                    |       TLS:           __thread                                                                                                                                                                                                              |       SIGSEGV:       normal                                                                                                                                                                                                                
|       Notifications: epoll                                                                                                                                                                                                                 
|       Architecture:  amd64                                                                                                                                                                                                                 
|       Disabled:      none                                                                                                                                                                                                                  
|       Misc:          softdebug                                                                                                                                                                                                             
|       Interpreter:   yes                                                                                                                                                                                                                   
|       LLVM:          supported, not enabled.                                                                                                                                                                                               
|       Suspend:       hybrid                                                                                                                                                                                                                
|       GC:            sgen (concurrent by default)                                                                                                                                                                                          
| + downloadMSBuildForMono                                                                                                                                                                                                                   
| + '[' '!' -e /build/tmp/work/corei7-64-poky-linux/msbuild/15.4-r0/git/packages/msbuild/MSBuild.exe ']'                                                                                                                                     
| + mkdir -p /build/tmp/work/corei7-64-poky-linux/msbuild/15.4-r0/git/packages                                                                                                                                                               
| + echo '** Downloading MSBUILD from https://github.com/Microsoft/msbuild/releases/download/mono-hosted-msbuild-v0.03/mono_msbuild_d25dd923839404bd64cc63f420e75acf96fc75c4.zip'                                                            
| ** Downloading MSBUILD from https://github.com/Microsoft/msbuild/releases/download/mono-hosted-msbuild-v0.03/mono_msbuild_d25dd923839404bd64cc63f420e75acf96fc75c4.zip                                                                     
| + curl -sL -o /build/tmp/work/corei7-64-poky-linux/msbuild/15.4-r0/git/packages/msbuild.zip https://github.com/Microsoft/msbuild/releases/download/mono-hosted-msbuild-v0.03/mono_msbuild_d25dd923839404bd64cc63f420e75acf96fc75c4.zip     
| WARNING: exit code 77 from a shell command.                                                                                                                                                                                                
| ERROR: Execution of '/build/tmp/work/corei7-64-poky-linux/msbuild/15.4-r0/temp/run.do_compile.433' failed with exit code 77
ERROR: Task (/work/meta-mono/recipes-mono/msbuild/msbuild_15.4.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2082 tasks of which 2080 didn't need to be rerun and 1 failed.
```

zboszor, 2022-05-16:
You are trying to build version 15.4. That has been retired by Microsoft to the point that even the original download sites (e.g. dotnet.myget.org) the build framework of that msbuild version was using were rearranged beyond recognition.
I had to move to msbuild 16.6. But to do that, I also had to upgrade to Mono 6.x.

Also, I just got a build error with 16.6 that popped up in Yocto 3.3 Hardknott, the error is:
```
|   It was not possible to find any installed .NET Core SDKs
|   Did you mean to run .NET Core SDK commands? Install a .NET Core SDK from:
|       https://aka.ms/dotnet-download
|   It was not possible to find any installed .NET Core SDKs
|   Did you mean to run .NET Core SDK commands? Install a .NET Core SDK from:
|       https://aka.ms/dotnet-download
| Retrying 'FindPackagesByIdAsync' for source 'https://api.nuget.org/v3-flatcontainer/nuget.build.tasks/index.json'.
| No such host is known
| Retrying 'FindPackagesByIdAsync' for source 'https://api.nuget.org/v3-flatcontainer/nuget.build.tasks/index.json'.
| No such host is known
| Retrying 'FindPackagesByIdAsync' for source 'https://api.nuget.org/v3-flatcontainer/microsoft.netframework.referenceassemblies.net472/index.json'.
| No such host is known
| Retrying 'FindPackagesByIdAsync' for source 'https://api.nuget.org/v3-flatcontainer/microsoft.netframework.referenceassemblies.net472/index.json'.
| No such host is known
| .../tmp-sicom-glibc/work/x86_64-linux/msbuild-native/16.6-r0/git/artifacts/mono-msbuild/NuGet.targets(123,5): error : Failed to retrieve information about 'NuGet.Build.Tasks' from remote source 'https://api.nuget.org/v3-flatcontainer/nuget.build.tasks/index.json'. [.../tmp-sicom-glibc/work/x86_64-linux/msbuild-native/16.6-r0/git/mono/build/sdks_and_nugets/nuget-build-tasks/nuget-build-tasks.csproj]
| .../tmp-sicom-glibc/work/x86_64-linux/msbuild-native/16.6-r0/git/artifacts/mono-msbuild/NuGet.targets(123,5): error :   No such host is known [.../tmp-sicom-glibc/work/x86_64-linux/msbuild-native/16.6-r0/git/mono/build/sdks_and_nugets/nuget-build-tasks/nuget-build-tasks.csproj]
```
I will try to upgrade the recipe to 16.10.1, maybe it will solve the problem.

zboszor, 2022-05-17:
It seems my problem was a temporary DNS outage. msbuild 16.6 can be built just fine.

ajlennon, 2023-04-03:
Closing this

closed 2023-04-03 by ajlennon