DynamicDevices/meta-mono #124: dotNet 5 not building in master

ajlennon, 2022-02-26:
Any thoguhts @CTerasaGuF ?

```
ERROR: dotnet-5.0.300-r0 do_package_qa: QA Issue: /usr/share/dotnet/shared/Microsoft.NETCore.App/5.0.6/libcoreclrtraceptprovider.so contained in package dotnet requires liblttng-ust.so.0()(64bit), but no providers found in RDEPENDS:dotnet? [file-rdeps]
```

ajlennon, 2022-02-26:
Looks like this is being discussed here

https://github.com/dotnet/runtime/issues/9494

ajlennon, 2022-03-31:
Yeah I was just looking at that. Can't quite see why we're seeing this problem with .0 versus .1

karthago1, 2022-05-01:
@ajlennon it seems that dotnet 5 is still not buildable in the master branch. should we just drop the dotnet 5 recipe in the master branch? Microsoft will end the support on [May 10, 2022](https://dotnet.microsoft.com/en-us/platform/support/policy/dotnet-core)

(I hope future releases of dotnet will switch to newer version of lttng-ust.)

ajlennon, 2022-05-01:
Yes that seems sensible to me?

ajlennon, 2022-07-18:
Closing this as we're on DotNet 5 now

closed 2022-07-18 by ajlennon