DynamicDevices/meta-mono #125: add dotnet 6 recipe

karthago1, 2022-03-30:
Hello !

Second PR :) ,

I renamed dotnet_5.0.300.bb to dotnet_6.0.201.bb and changed all URLs and checksums

both dotnet versions have the same folder structure. so the install logic of dotnet 5 should be valid for dotnet 6

I tested only `arm64`.  `x64` will be soon tested
(`arm` will not be tested, but there is no reason why it shouldn't work)

it will be nice if you accept this PR

karthago1, 2022-03-30:
sure I will try to modify the CI in the same PR

ajlennon, 2022-03-30:
Lovely thanks

karthago1, 2022-03-31:
hello @ajlennon 

I have to thank you too, since I learned new things like the yocto imagetest.
it is still work in progress. I will write here as soon as I finished it.

karthago1, 2022-03-31:
@ajlennon

as I said I just added an insane skip to ignore the lttng-ust dependency. I think it is safe to do it for now. but we should remove it as soon as dotnet runtime switches to use the new lib.

for now I have a problem with dotnet-helloworld_1.0.bb. to compile the source code, the command `dotnet build` is executed, which needs to access this url https://api.nuget.org/v3/index.json.

the command is working fine in my environment (with your docker image). but the CI seems to have no access to that file. may be you have an idea, what is going wrong?

`/__w/meta-mono/meta-mono/master/build/tmp/work/core2-64-poky-linux/dotnet-helloworld/1.0-r0/recipe-sysroot-native/usr/share/dotnet/sdk/6.0.201/NuGet.targets(130,5): error : Unable to load the service index for source https://api.nuget.org/v3/index.json. [/__w/meta-mono/meta-mono/master/build/tmp/work/core2-64-poky-linux/dotnet-helloworld/1.0-r0/src/dotnet-helloworld.csproj]`

ajlennon, 2022-03-31:
I tried that myself but then the rootfs failed to build for some reason?

closed 2022-04-26 by karthago1