DynamicDevices/meta-mono #147: AOT cross-compiling

klasyc, 2022-08-25:
In order to speed up my mono application on low-performance hardware, I would like to precompile its binaries and GAC using [mono Ahead-Of-Time compiler](https://www.mono-project.com/docs/advanced/aot/). I have already tried to run AOT compiler on the target device - it requires a bunch of dependencies and the compiling takes quite a long time. But the performance improvement was significant, and I would like to integrate AOT cross-compiling into Yocto build system.

Here I would like to discuss, how the final build system should work:

1. First, I need to build mono as a cross-compiler. For this purpose, I created a new `mono-cross` recipe which inherits the `cross` class. After [some problems](https://github.com/mono/mono/issues/20035), I successfully built the cross-compiler. Unfortunately, my recipe fails in `do_install` task because some of the files conflict with the same files provided by the `mono-native` recipe. Any suggestions how to deal with this problem? Maybe install the mono cross-compiler to some prefix? I also noticed that the `cross` class adds a prefix to all binaries, so the conflict appears only when installing some libraries - at this point I am not sure if that libraries differ from the native binaries provided by `mono-native` recipe. Maybe it would be sufficient just not try to install them again...

2. I need to find a way how to precompile GAC, which is generated by `mono` recipe. The first idea was just to create a new variant of `mono` recipe (e.g. `mono-aot`) which would execute my cross-compiler after compiling basic mono. But this solution would require to create a new variant for each `mono` version, which is hardly scalable. So the second idea is to modify `mono` recipe directly (or more exactly, one of its `.inc` files) and introduce a global variable which will turn the AOT feature on.

There are not many resources on this topic, but according to this [Stack Overflow post](https://stackoverflow.com/questions/51629444/mono-cross-platform-aot-ahead-of-time-compiling) it should be possible.

I will be thankful for your suggestions.

ajlennon, 2022-08-25:
Hi @klasyc. Great work. Interesting stuff.

I'm not sure. It feels like it lends itself to adding new tasks to the bitbake recipe, perhaps driven by a DISTRO_FEATURE setting?

This might be something you could ask on the Poky or Bitbake lists?

ajlennon, 2023-04-03:
Closing due to inactivity

closed 2023-04-03 by ajlennon

klasyc, 2023-04-03:
I don't want to leave my issue without resolution, so I would like to sum up what I have discovered. Finally, I got the AOT cross-compiling working. Here is the approach I ended up with (only in a nutshell because the journey was really long):

1. I created a new recipe `mono-cross-xxx` which boils the AOT cross-compiler. The recipe is quite similar to the `mono-native`, but uses different mono configuration.
2. I created a new include file `mono-aot-xxx.inc` which can be included to the `mono` recipes where the GAC precompiling is required. Basically, this file adds `do_install_append()` function which calls the AOT cross-compiler for libraries in GAC.
3. In order to use the `mono-aot-xxx.inc` file, I created `mono_6%.bbappend` append file, which simply includes the GAC precompiling to all mono 6 recipes.
4. I created a new class `mono-aot` which can be inherited by the user recipes which require the AOT cross-compiling. The class exposes `MONO_AOT` variable which contains the fully qualified name of the AOT cross-compiler and also the basic command-line arguments. The recipe has to define `do_compile()` function with the following content:
   ```
   do_compile () {
       ${MONO_AOT} -O=all ${S}/*.dll ${S}/*.exe
   }
   ```
The main issues were:
- In order to make the cross-compiled DLL files compatible with the embedded mono, both compilers (obviously) have to use the same configuration. Unfortunately, the mono cross-compiler does not support the same threading configurations as the embedded mono and does not comply about it. It took me much of time to realize that I have to change the embedded mono configuration first to make it possible to use the same configuration for both monos. Finally my configuration is:
   ```
   mono_cv_uscore=no --with-sigaltstack=no --with-mcs-docs=no --with-cross-offsets=${WORKDIR}/armv7hf-offsets.h --disable-boehm --enable-nls --with-tls=pthread --enable-hybrid-suspend=no --enable-cooperative-suspend=no
   ```
- In order to compile the cross-compiler mono, it is necessary to provide the "offsets file" which contains the sizes and alignments of the basic types and structures used in the generated byte code. Mono provides `offsets-generator` tool, which is a python script that compiles some mono headers using the target GCC compiler. Unfortunately, this tool has some problems:
   - I wasn't able to generate the offsets file without modifying the script. There is lack of documentation, and maybe I don't understand correctly how to use it. But maybe there are some bugs involved.
   - The tool is not fully compatible with my Yocto's GCC. I had to add some symlinks to make it working.
   - The `offsets-tool` is not part of the binary `mono` distribution, i.e. it is not present in the `mono-xxxx.tar.gz` archive downloaded by Yocto. It would have to be downloaded separately from Git.

   For the reasons above and also because I need only ARM architecture, I simply generated the offsets file outside the Yocto build system, and then I supply this offsets file as a "constant". I understand this is not the right Yocto way, but it works, and I was not able to devote more time to this issue...

Now the cross-compiling works and the performance benefit is really significant: my Gtk# application startup time reduced from 19 seconds to 12 seconds and the application works more fluently. On the other hand, the precompiled GAC and application DLLS take approximately 130 MB of the extra flash memory.

I made all the things above in my custom layer, and I am not going to create a pull request because:
- I am using EOL `morty` distribution.
- I don't like the way of the `offsets-tool` integration.

I hope that somebody will find my report useful :)

ajlennon, 2023-04-05:
That's really really useful - thanks for taking your time to put these comments down for the community @klasyc