DynamicDevices/meta-mono #148: mono-native fails to build when using meta-frss

mferland, 2022-09-23:


GamerBroGebauer, 2022-09-26:
Hi i got the same behavior with the following error:

``` ERROR: mono-native-6.12.0.161-r0 do_prepare_recipe_sysroot: Manifest /workdir/build/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-zlib.populate_sysroot not found in phytec_mx8mp cortexa53 armv8a-crc armv8a aarch64 allarch x86_64_x86_64-nativesdk (variant '')?
ERROR: Logfile of failure stored in: /workdir/build/tmp/work/x86_64-linux/mono-native/6.12.0.161-r0/temp/log.do_prepare_recipe_sysroot.1610929
ERROR: Task (/workdir/sources/meta-mono/recipes-mono/mono/mono-native_6.12.0.161.bb:do_prepare_recipe_sysroot) failed with exit code '1'
```

Is there a workaround or fix available?

GamerBroGebauer, 2022-09-26:
> I'm currently trying to build mono-native (along with mono) using the imx8mn-ddr4-evk machine defined in the meta-freescale layer.
> 
> This resulted in the following build error:
> 
> ```
> ERROR: mono-native-6.12.0.161-r0 do_prepare_recipe_sysroot: Manifest /workdir/build/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-zlib.populate_sysroot not found in imx8mn_ddr4_evk cortexa53 armv8a-crc armv8a aarch64 allarch x86_64_x86_64-nativesdk (variant '')?
> ERROR: Logfile of failure stored in: /workdir/build/tmp/work/x86_64-linux/mono-native/6.12.0.161-r0/temp/log.do_prepare_recipe_sysroot.1588124
> ERROR: Task (/workdir/yocto/meta-mono/recipes-mono/mono/mono-native_6.12.0.161.bb:do_prepare_recipe_sysroot) failed with exit code '1'
> ```
> 
> I did _not_ observe this error when building with qemuarm. Looking at the DEPENDS variable for the mono-native package I noticed the following:
> 
> ```
> $ bitbake -e mono-native | grep ^DEPENDS
> DEPENDS="pkgconfig-native autoconf-native automake-native libtool-native   gettext-minimal-native zlib libatomic-ops  cmake-native python3-native  libgdiplus-native libgdiplus-native"
> ```
> 
> Somehow, the `-native` build depends on `zlib` and `libatomic-ops` which is kinda strange (should've been zlib-native and libatomic-ops-native). This is when I noticed that the mono-6.xx.inc file adds a DEPENDS on zlib and libatomic-ops but does not define `BBCLASSEXTEND = "native"` (which I think is the culprit here).
> 
> Once `BBCLASSEXTEND = "native"` is defined in mono-6.xx.inc, DEPENDS now report the following:
> 
> ```
> DEPENDS="pkgconfig-native autoconf-native automake-native libtool-native   gettext-minimal-native zlib libatomic-ops  cmake-native python3-native  libgdiplus-native libgdiplus-native"
> ```
> 
> which make more sense.
> 
> I would like to know if this makes sense for you, if it does, I'll provide a patch.
> 
> I am using the following revisions (all kirkstone):
> 
> ```
> meta                 
> meta-poky            
> meta-yocto-bsp       = "mfe/kirkstone:d64bef1c7d713b92a51228e5ade945835e5a94a4"
> meta-oe              = "mfe/kirkstone:05dcac98473402d87e0af73bbc2c5a6a840abe93"
> meta-freescale       = "mfe/kirkstone:3cd0c61e6051d52c6068bf5d0a3c4ef5386e4c90"
> meta-mono            = "mfe/kirkstone:0d49d37aaa66356db55e3beb86f33426a50629e2"
> ```
> 
> Thanks!

Hi @mferland i have added `BBCLASS ='native'` to `mono-6.xx.inc` locally and mono-native builds successfully in my case for currently 27m.

GamerBroGebauer, 2022-09-26:
> > I'm currently trying to build mono-native (along with mono) using the imx8mn-ddr4-evk machine defined in the meta-freescale layer.
> > This resulted in the following build error:
> > ```
> > ERROR: mono-native-6.12.0.161-r0 do_prepare_recipe_sysroot: Manifest /workdir/build/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-zlib.populate_sysroot not found in imx8mn_ddr4_evk cortexa53 armv8a-crc armv8a aarch64 allarch x86_64_x86_64-nativesdk (variant '')?
> > ERROR: Logfile of failure stored in: /workdir/build/tmp/work/x86_64-linux/mono-native/6.12.0.161-r0/temp/log.do_prepare_recipe_sysroot.1588124
> > ERROR: Task (/workdir/yocto/meta-mono/recipes-mono/mono/mono-native_6.12.0.161.bb:do_prepare_recipe_sysroot) failed with exit code '1'
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > I did _not_ observe this error when building with qemuarm. Looking at the DEPENDS variable for the mono-native package I noticed the following:
> > ```
> > $ bitbake -e mono-native | grep ^DEPENDS
> > DEPENDS="pkgconfig-native autoconf-native automake-native libtool-native   gettext-minimal-native zlib libatomic-ops  cmake-native python3-native  libgdiplus-native libgdiplus-native"
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Somehow, the `-native` build depends on `zlib` and `libatomic-ops` which is kinda strange (should've been zlib-native and libatomic-ops-native). This is when I noticed that the mono-6.xx.inc file adds a DEPENDS on zlib and libatomic-ops but does not define `BBCLASSEXTEND = "native"` (which I think is the culprit here).
> > Once `BBCLASSEXTEND = "native"` is defined in mono-6.xx.inc, DEPENDS now report the following:
> > ```
> > DEPENDS="pkgconfig-native autoconf-native automake-native libtool-native   gettext-minimal-native zlib libatomic-ops  cmake-native python3-native  libgdiplus-native libgdiplus-native"
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > which make more sense.
> > I would like to know if this makes sense for you, if it does, I'll provide a patch.
> > I am using the following revisions (all kirkstone):
> > ```
> > meta                 
> > meta-poky            
> > meta-yocto-bsp       = "mfe/kirkstone:d64bef1c7d713b92a51228e5ade945835e5a94a4"
> > meta-oe              = "mfe/kirkstone:05dcac98473402d87e0af73bbc2c5a6a840abe93"
> > meta-freescale       = "mfe/kirkstone:3cd0c61e6051d52c6068bf5d0a3c4ef5386e4c90"
> > meta-mono            = "mfe/kirkstone:0d49d37aaa66356db55e3beb86f33426a50629e2"
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Thanks!
> 
> Hi @mferland i have added `BBCLASSEXTEND ='native'` to `mono-6.xx.inc` locally and mono-native builds currently in my case for 27minutes without any error.

Fix Works. Mono was build successfully. In my case you can bring the patch.

ajlennon, 2022-12-20:
Closing this as fixed

closed 2022-12-20 by ajlennon