DynamicDevices/meta-mono #150: Possible host contamination when using nuget

mferland, 2022-10-03:
I'm not an expert on .NET, but looking at the log.do_compile of one of our package, I noticed the following:
```
  NuGet Config files used:
      /home/pokyuser/.config/NuGet/NuGet.Config
```
This could lead to issues, especially when not using a docker environment. The build could pickup host configurations through this file (aka host contamination).

mferland, 2023-04-03:
Did some quick tests but wasn't able to reproduce.

ajlennon, 2023-04-04:
ok thanks for that - I'll close this for now then :)

CrazyGecko, 2025-05-08:
A quick update, if someone have troubles with that:
On my side, I was able to reproduce this behavior and it looks like it still exist. Also NuGet creates the file if it don't exist in this folder. There is the possibility to give a specific config file path to NuGet, but it sill additionally consumes and creates the file in the _home_ folder. However, I would say this is an issue of NuGet and can't be fixed here. The only way I was able to escape the host contamination was to change the `HOME` variable, which then led to other errors.