DynamicDevices/meta-mono #151: kirkstone: gtk-sharp3 unable to find revision in branch and Nothing Provides libglade

GamerBroGebauer, 2022-10-11:
I try to add the recipe gtk-sharp3 into my image and i run into the following two errors.

``` 
ERROR: gtk-sharp3-2.99.4-r0 do_fetch: Fetcher failure: Unable to find revision 9a72bb67fff7e4845b7bb430a608282668c3e4da in branch master even from upstream
ERROR: gtk-sharp3-2.99.4-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'git://github.com/mono/gtk-sharp.git;branch=master')
```

For this error i observed that the master branch was renamed to main in [source repository](https://github.com/mono/gtk-sharp)

After I had fixed the first error by changing the  `branch=master` to `branch=main` statement in gtk-sharp3_2.99.4.bb recipe i've got the following error.

```
Nothing PROVIDES 'libglade' (but /yocto/sources/meta-mono/recipes-mono/gtk-sharp/gtk-sharp3_2.99.4.bb DEPENDS on or otherwise requires it). Close matches:
  glade
  libgudev
  vblade
```

I've changed the statements `DEPENDS += " gtk+3 atk pango cairo glib-2.0 libglade mono"` to `DEPENDS += " gtk+3 atk pango cairo glib-2.0 glade mono"` inside the gtk-sharp3_2.99.4.bb recipe but this was unbuildable with following error message.

```
function 'mono_profiler_set_events'; did you mean 'mono_profiler_startup'? [-Wimplicit-function-declaration] 
|    96 |         mono_profiler_set_events (MONO_PROFILE_ENTER_LEAVE); 
|       |         ^~~~~~~~~~~~~~~~~~~~~~~~ 
|       |         mono_profiler_startup 
| ../../../../git/gtk/gui-thread-check/profiler/gui-thread-check.c:96:9: warning: nested extern declaration of 'mono_profiler_set_events' [-Wnested-externs] 
| ../../../../git/gtk/gui-thread-check/profiler/gui-thread-check.c:96:35: error: 'MONO_PROFILE_ENTER_LEAVE' undeclared (first use in this function) 
|    96 |         mono_profiler_set_events (MONO_PROFILE_ENTER_LEAVE); 
|       |                                   ^~~~~~~~~~~~~~~~~~~~~~~~ 
| ../../../../git/gtk/gui-thread-check/profiler/gui-thread-check.c:96:35: note: each undeclared identifier is reported only once for each function it appears in 
```

Has somebody had the same issue? Is there currently a solution available for the second error?

If i use gtk-sharp recipe instead of the gtk-sharp3 it builds without of problems. Is gtk-sharp and gtk-sharp3 the same or can somebody explane me the differences please?

If  the gtk-sharp3 is buildable i will bring in all necessary  changes.

electroScorpion, 2023-04-03:
Hi @ajlennon for me not because i am using now the gtk-sharp package but it would be great to understand the background why this happens and how to fix it.

ajlennon, 2023-04-03:
Will take a look

Livius90, 2024-06-04:
- Patch for 'MONO_PROFILE_ENTER_LEAVE' undeclared issue:
https://github.com/mono/gtk-sharp/pull/266
- libglade is removed from meta-openembedded because it was a dead project since 2009, it was available until Yocto rocko release.
https://layers.openembedded.org/layerindex/recipe/70296/
https://git.openembedded.org/meta-openembedded/commit/meta-oe/recipes-gnome?id=9dcf44fc15b565db1ba654333218bd4b711d01ca