DynamicDevices/meta-mono #155: Update dotnet 6.0.X recipe?

Livius90, 2022-11-06:
Is there any plan to update dotnet recipe to the latest version in kirkstone and master branch? Now the latest is [.NET 6.0.10 - October 11, 2022](https://github.com/dotnet/core/blob/main/release-notes/6.0/6.0.10/6.0.10.md).

ajlennon, 2022-11-07:
Yes - just trying to work out why the Kirksstone CI isn't building at the moment...

https://github.com/DynamicDevices/meta-mono/actions/runs/3377736232/jobs/5661399849

Livius90, 2022-11-08:
> Yes - just trying to work out why the Kirksstone CI isn't building at the moment...
> 
> https://github.com/DynamicDevices/meta-mono/actions/runs/3377736232/jobs/5661399849

In gentoo forum there is a same "No rule to make target" issue. Solution was to "make clean". In bitbake you can try -c cleanall for linux-yocto recipe or you have a kernel config problem.
https://forums.gentoo.org/viewtopic-t-1094794-start-0.html

ajlennon, 2022-11-08:
Yeah it's all been a bit of a nightmare. Spent the day on it today having bashed away at it numbers of times recently. I've updated the docker build container to 22.04 and pulled in some bits that they've deprecated so kirkstone head is now building (yay)

The current build container is specified here - https://github.com/DynamicDevices/yocto-ci-build

Kirkstone build and test is now passing - https://github.com/DynamicDevices/meta-mono/actions/runs/3419114336

So we're now back onto a stable basis for accepting PRs...

ajlennon, 2022-11-08:
Would you be interested in looking at creating a PR to bring dotNet up to date @Livius90 ?

Livius90, 2022-11-08:
> Would you be interested in looking at creating a PR to bring dotNet up to date @Livius90 ?

https://github.com/DynamicDevices/meta-mono/pull/157

ajlennon, 2022-11-08:
Merged so will close - many thanks

closed 2022-11-08 by ajlennon