DynamicDevices/meta-mono #168: master build against langdale is failing on python3-pythonnet

ajlennon, 2023-04-03:
@see https://github.com/DynamicDevices/meta-mono/actions/runs/4587928068/jobs/8104065448

Livius90, 2023-04-03:
Hi,
I see what should be the problem. There is a new bbclass in Yocto to install python packages which can support .toml format instead of use legacy setup.py. Kirkstone was the first Yocto release which could support it via `inherit python_flit_core`. But when i tried to use it for pythonnet compiling and installing it was buggy and somewhy not all of .py files were deployed to target rootfs. My workaround was to make a custom [do_install:prepend()](https://github.com/DynamicDevices/meta-mono/blob/master/recipes-python/python3-pythonnet/python3-pythonnet.bb#L65) to intall missing things manually in the recipe.

Probably the following manual install is not needed any more, because it may have been fixed in Langdale: `install -m 644 ${S}/clr.py    ${D}${PYTHON_SITEPACKAGES_DIR}`
https://github.com/DynamicDevices/meta-mono/blob/master/recipes-python/python3-pythonnet/python3-pythonnet.bb#L72

Also, useful to be tested that the `do_compile:prepend()` is necessery or not anymore. In normal operation .toml python project should done the dotnet compiling automaticly.
https://github.com/DynamicDevices/meta-mono/blob/master/recipes-python/python3-pythonnet/python3-pythonnet.bb#L60

By the way i am planning to make a refactoring for pythonnet and clr-loader recipe, because these recipes fetch the source from original GitHub sites, but usually it is much better and faster to download their released source from [pypi.org](https://pypi.org/) for yocto recipes.

ajlennon, 2023-04-04:
Great thanks!

closed 2023-04-06 by ajlennon