DynamicDevices/meta-mono #172: Hardknott branch fails to build

ajlennon, 2023-04-06:
Seems to be cmake and GLIBC

[image]

ref: https://github.com/DynamicDevices/meta-mono/actions/runs/4628040614/jobs/8186643790#step:7:1245

ajlennon, 2023-04-06:
Any thoughts @Livius90 @zboszor @karthago1 ?

zboszor, 2023-04-06:
What is the host OS?

Fedora 33 has no problem rebuilding honister and hardknott, but at least hardknott didn't build on Fedora 36 and up as far as I remember. I don't remember the exact issues, though.

A new problem would occur with mickledore upgraded to Mesa 23.0.0 and Ubuntu 18.0.4 LTS and others still having GCC 7.x can't build mesa-native because Mesa officially needs GCC 8.0 or newer.

ajlennon, 2023-04-06:
We should be on the "latest" docker container which should be what all the recent builds use

ref: https://github.com/DynamicDevices/meta-mono/blob/hardknott/.github/workflows/CI_github.yml#L14

zboszor, 2023-04-06:
As for the problem in question: I can see "uninative" in the paths.
The uninative recipe is constantly updated in Yocto for "newer GLIBC support".
This recipe seems to do something very low level and has an upper bound on supported GLIBC versions.
If you are on a newer host OS with a relatively recent GLIBC version, older Yocto versions may fail to build because of this.

ajlennon, 2023-04-06:
Hmmm. Most odd. As I say the build is containerised and the container should be the same that is being used for other builds that are currently working. Maybe I'll blow out the build tree and start it from scratch then

Livius90, 2023-04-06:
Some years ago my personal expereince was that Yocto sumo release became un-buildable in the newest Ubuntu 20.04 because of any similar glibc copmatibility issue. I think, it is a really common issue which means sooner or later old and legacy Yocto releases will be totally not buildable in newest Linux disrtibutions.

Livius90, 2023-07-27:
What about this issue? In my build PC i updated my Ubuntu to 22.04, Yocto sumo branch is not buildable anymore in this version of Ubuntu, unfortunately. :worried:

ajlennon, 2023-07-27:
😢 any PRs to fix it ?

zboszor, 2023-07-28:
> Some years ago my personal expereince was that Yocto sumo release became un-buildable in the newest Ubuntu 20.04 because of any similar glibc copmatibility issue. I think, it is a really common [...]

Exactly. I think this didn't change.

Sumo may still be built on CentOS 7 but in order to keep downloads from failing if possible, don't throw away the original downloads directory.

But downloads from git repos may still fail and you need to patch some things in Yocto recipes or add bbappend files to fix things like git repos renaming "master" to "main" and github requiring https:// access for a while. Some recipes in Sumo used the default unauthenticated plaintext git protocol and github disabled that.

ajlennon, 2023-07-28:
>Exactly. I think this didn't change.

True - it's for reasons including this I build releases in a defined Docker container in the CI

ajlennon, 2024-02-12:
I'm going to close this as `hardknott` is EOL and unless somebody wants to put time into it I won't

closed 2024-02-12 by ajlennon