DynamicDevices/meta-mono #205: mono-xsp do_configure recipe fails to build kirkstone

Lance-Kraakman, 2024-02-06:
Hi guys, I would love some help with this.

mono install fine, mono-xsp do_configure fails with "pkg-config" not 

[log.do_configure.log](https://github.com/DynamicDevices/meta-mono/files/14186006/log.do_configure.log)

```
Importing into BTLS user store:
I already trust 127, your new list has 127
Import process completed.
Error: pkg-config is not installed, and is required to configure 
Configure failed. The contents of all config.log files follows to aid debugging
ERROR:  failed
WARNING: exit code 1 from a shell command.
```

Environment variables at time of do_configure task are: (I appended the recipe to print out these environment vars)
PKG_CONFIG: 
PKG_CONFIG_PATH: /home/build/yocto-sources/build-conf/tmp/work/corei7-64-poky-linux/mono-xsp/git-r0/recipe-sysroot/usr/lib/pkgconfig:/home/build/yocto-sources/build-conf/tmp/work/corei7-64-poky-linux/mono-xsp/git-r0/recipe-sysroot/usr/share/pkgconfig

I have tried:

    - inherit pkgconfig 
    - exporting pkg-config binaries
    - made sure pkgconfig is installed on host
    
I am on the latest kirkston versions for all of my layers.
Thanks you :)

Lance-Kraakman, 2024-02-29:
:(

ajlennon, 2024-03-01:
Hmmm. We could have a look at if this is building within the meta-mono Kirkstone CI?

Maybe add `mono-xsp` to `test-image.mono.bb`, create a PR, we'll run the CI and see what happens?

https://github.com/DynamicDevices/meta-mono/blob/master/recipes-mono/images/test-image-mono.bb

ajlennon, 2024-03-03:
ok so we do see the failure with the CI....

https://github.com/DynamicDevices/meta-mono/actions/runs/8129076040/job/22215890514?pr=216#step:8:14412

Lance-Kraakman, 2024-03-03:
> ok so we do see the failure with the CI....
> 
> https://github.com/DynamicDevices/meta-mono/actions/runs/8129076040/job/22215890514?pr=216#step:8:14412

Bugger, can you get your hands on the log file from the machine after failure?
Adding pkg-config to the HOSTTOOLS environment variable's gets it "a little bit further along" from memory, but still fails.

Are you able to look at this? I don't mind trying things out if you have suggestions.

ajlennon, 2024-06-08:
Any progress here? If not I'll close due to inactivity

Lance-Kraakman, 2024-06-09:
No sorry, we are porting to dotnet instead.

closed 2024-06-10 by ajlennon