DynamicDevices/meta-mono #219: Mono runtime error: Could not load file or assembly Microsoft.VisualBasic, Version=8.0.0.0

obaboli, 2024-03-06:
Hi! I'm using the mono layer to run a closed source .NET exe. It looks to be using .NETFramework Version 4.0. I'm able to run one of the CLIs but the second results in a runtime error below:

```
ERROR: System.IO.FileNotFoundException: Could not load file or assembly 'Microsoft.VisualBasic, Version=8.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a' or one of its dependencies.
File name: 'Microsoft.VisualBasic, Version=8.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'
```

We currently use yocto kirkstone and include only "mono" in our IMAGE_INSTALL:append. Are there any additional recipes to get this assembly in our yocto build?

ajlennon, 2024-03-08:
Hi @obaboli. Interesting question. I may be wildly incorrect but it looks as though we may need support for this

https://github.com/mono/mono-basic

obaboli, 2024-03-13:
> Hi @obaboli. Interesting question. I may be wildly incorrect but it looks as though we may need support for this
> 
> https://github.com/mono/mono-basic
> 
> [Edit: Off the top of my head we don't currently have a recipe for this...]

Makes sense. 

I did notice there is a mono-basic recipe inside recipes-mono.  Tried to build it and got some errors with kirkstone. 

```
aarch64-oe-linux/mono-basic/4.0-r0/mono-basic-4.0/vbnc/vbnc/source/ConditionalCompilation/ConditionalExpression.vb (284,50) : error VBNC99999: Internal compiler error close to this location
| vbnc : Command line : error VBNC99999: Unexpected error: The method or operation is not implemented.
|   at vbnc.CecilHelper.InflateType (Mono.Cecil.TypeReference original, Mono.Collections.Generic.Collection`1[T] parameters, Mono.Collections.Generic.Collection`1[T] arguments) [0x003f1] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/General/CecilHelper.vb:422
|   at vbnc.CecilHelper.GetCorrectMember (Mono.Cecil.MethodDefinition Member, Mono.Cecil.TypeReference Type, System.Boolean Emittable) [0x000bd] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/General/CecilHelper.vb:636
|   at vbnc.CecilHelper.GetCorrectMember (Mono.Cecil.MethodReference Member, Mono.Cecil.TypeReference Type, System.Boolean Emittable) [0x0001b] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/General/CecilHelper.vb:520
|   at vbnc.CecilHelper.GetMembers (Mono.Cecil.TypeReference Type) [0x00218] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/General/CecilHelper.vb:99
|   at vbnc.MemberCache.ClearAll () [0x0010f] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/General/MemberCache.vb:122
|   at vbnc.MemberCache..ctor (vbnc.Compiler Compiler, Mono.Cecil.TypeReference Type) [0x00059] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/General/MemberCache.vb:69
|   at vbnc.TypeManager.GetCache (Mono.Cecil.TypeReference Type) [0x00028] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/General/TypeManager.vb:87
|   at vbnc.Helper.GetOperators (vbnc.Compiler Compiler, System.String[] Names, Mono.Cecil.TypeReference Type) [0x0000b] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/General/Helper.vb:1640
|   at vbnc.Helper.GetConversionOperators (vbnc.Compiler Compiler, System.String[] Names, Mono.Cecil.TypeReference Type, Mono.Cecil.TypeReference ReturnType) [0x0002e] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/General/Helper.vb:1612
|   at vbnc.Helper.GetWideningConversionOperators (vbnc.Compiler Compiler, Mono.Cecil.TypeReference Type, Mono.Cecil.TypeReference ReturnType) [0x00000] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/General/Helper.vb:1628
|   at vbnc.Helper.IsConvertible (vbnc.ParsedObject Parent, vbnc.Expression FromExpression, Mono.Cecil.TypeReference FromType, Mono.Cecil.TypeReference DestinationType, System.Boolean CreateConversionExpression, vbnc.Expression& convExpr, System.Boolean ShowError, System.Nullable`1[T] isStrict, System.Boolean considerConstantExpressions) [0x037af] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/General/Helper.vb:3662
|   at vbnc.MemberCandidate.IsApplicable (System.Collections.Generic.List`1[T] error_lines) [0x00d4c] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/Members/MethodResolver.vb:1619
|   at vbnc.MethodResolver.RemoveInapplicable (System.Collections.Generic.List`1[T] error_lines) [0x0006a] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/Members/MethodResolver.vb:584
|   at vbnc.MethodResolver.ResolveInternal () [0x00335] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/Members/MethodResolver.vb:242
|   at vbnc.MethodResolver.Resolve () [0x00008] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/Members/MethodResolver.vb:162
|   at vbnc.MethodGroupClassification.ResolveGroup (vbnc.ArgumentList SourceParameters, System.Boolean ShowErrors, System.Boolean CanBeLateBound) [0x000e2] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/Expressions/Classifications/MethodGroupClassification.vb:403
|   at vbnc.InvocationOrIndexExpression.ResolveMethodInvocation () [0x0013a] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/Expressions/InvocationOrIndexExpression.vb:514
|   at vbnc.InvocationOrIndexExpression.ResolveExpressionInternal (vbnc.ResolveInfo Info) [0x00368] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/Expressions/InvocationOrIndexExpression.vb:238
|   at vbnc.Expression.ResolveExpression (vbnc.ResolveInfo ResolveInfo) [0x0007a] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/Expressions/Expression.vb:225
|   at vbnc.AssignmentStatement.ResolveStatement (vbnc.ResolveInfo Info) [0x0002f] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/Statements/AssignStatements/AssignmentStatement.vb:82
|   at vbnc.CodeBlock.ResolveCode (vbnc.ResolveInfo Info) [0x000c6] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/Code/CodeBlock.vb:570
|   at vbnc.IfStatement.ResolveStatement (vbnc.ResolveInfo Info) [0x0002f] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/Statements/IfStatement.vb:117
|   at vbnc.CodeBlock.ResolveCode (vbnc.ResolveInfo Info) [0x000c6] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/Code/CodeBlock.vb:570
|   at vbnc.IfStatement.ResolveStatement (vbnc.ResolveInfo Info) [0x0006a] in /home/directhex/Projects/mono-basic/vbnc/vbnc/source/Statements/IfStatement.vb:118
[... cut ...]
| make[2]: Leaving directory '/home/turbox/workspace/sourcecode/turbox-c8550-le1.0-dev.release.FC.r001003/LE.PRODUCT.2.1.r1/apps_proc/build-qti-distro-rb-debug/tmp-glibc/work/aarch64-oe-linux/mono-basic/4.0-r0/mono-basic-4.0/vbnc/vbnc'
| make[2]: *** [../../build/rules.make:122: do-all] Error 2
| make[1]: Leaving directory '/home/turbox/workspace/sourcecode/turbox-c8550-le1.0-dev.release.FC.r001003/LE.PRODUCT.2.1.r1/apps_proc/build-qti-distro-rb-debug/tmp-glibc/work/aarch64-oe-linux/mono-basic/4.0-r0/mono-basic-4.0/vbnc'
| make[1]: *** [../build/rules.make:134: all-recursive] Error 1
| make: *** [build/rules.make:134: all-recursive] Error 1
| ERROR: oe_runmake failed
```

ajlennon, 2024-06-08:
Any progress here? If not I'll close due to inactivity