DynamicDevices/meta-mono #232: CI keeps blocking

ajlennon, 2024-10-04:
Never used to do this - feels as though it's related to the latest Yocto releases? A reboot seems to unblock it but needs looking into

zboszor, 2024-10-07:
I have had an issue (bitbake instances get stuck even when a build finishes successfully) when I first switched to Fedora 40 from Fedora 39. It was partly caused by a bug in early Python 3.12.x versions which was later fixed in 3.12.3 or so. The other part was this bugfix in `bitbake` which should be included in nanbield or later, and it's also backported to at least the kirkstone version of `bitbake`:
```
commit d01d684a0f6398270fe35ed59b7d28f3fd9b7e41
Author: Joshua Watt <JPEWhacker@gmail.com>
Date:   Fri Nov 3 08:26:21 2023 -0600

    asyncrpc: Add context manager API
    
    Adds context manager API for the asyncrcp client class which allow
    writing code that will automatically close the connection like so:
    
        with hashserv.create_client(address) as client:
           ...
    
    Rework the bitbake-hashclient tool and PR server to use this new API to
    fix warnings about unclosed event loops when exiting
    
    Signed-off-by: Joshua Watt <JPEWhacker@gmail.com>
    Signed-off-by: Richard Purdie <richard.purdie@linuxfoundation.org>
```

ajlennon, 2024-10-07:
Very interesting thanks !!!

ajlennon, 2024-10-07:
I've just checked the container and it's running Ubuntu 22.04 with Python 3.10.6

@see https://github.com/DynamicDevices/yocto-ci-build/blob/main/Dockerfile

Maybe I'll try rebuilding the container and see if that helps anything

ajlennon, 2024-10-07:
I seem to be installing Python 2.7.18 as well although I can't remember why

ajlennon, 2024-10-07:
ok I've pushed a new build of the build container image up to Docker Hub. Will see if that makes any difference...

ajlennon, 2024-10-09:
Seems better since I pushed that container rebuild. Hasn't blocked... will keep on keeping an eye on..

ajlennon, 2024-10-09:
Haven't seen any more problems over a few builds so will close this for now