DynamicDevices/meta-mono #238: Mono packages don't build properly with multilib

niraami, 2024-10-17:
When building mono for a machine with multilib (for x32) enabled, most packages, specifically `mono-gac` and `mono-libs-4.5` don't get built, leading to errors like this when the target image is in it's `do_rootfs` step:
```
nothing provides mono-gac needed by mono-4.6.1.5-r0.corei7-64
nothing provides mono-libs-4.5 needed by mono-4.6.1.5-r0.corei7-64
```

The reason is that, in the `:append` to `do_install` in [`mono-base.inc`](recipes-mono/mono/mono-base.inc) looks for libs in `${STAGING_DIR_NATIVE}${libdir}`, but when multilib is enabled, then `libdir` = `/usr/lib64`, but the libs are in `/usr/lib`.
Permalink:
https://github.com/DynamicDevices/meta-mono/blob/a4d4550223f5d4437bb25b94e1f9075ff69cf9ba/recipes-mono/mono/mono-base.inc#L37-L44

I've temporarily fixed this by replacing `libdir` with `base_libdir` (`libdir_native` would also work), but maybe this fix should instead target the part of the build system that actually installs the libs (in `mono-native`), so that it uses `libdir` and not `/usr/lib`..?

Also, we're using Kirkstone, I don't know if this affects other versions as well. Our old dunfell built just fine in the past with multilib, so I'm unsure when this issue got introduced.

You can test this by adding the following to your `<MACHINE>.conf` (thought you might need a different defaulttune depending on the target you're building for):
```
# Enable 32-bit multilib support
require conf/multilib.conf
DEFAULTTUNE = "corei7-64"
MULTILIBS = "multilib:lib32"
DEFAULTTUNE:virtclass-multilib-lib32 = "x86"
```

ajlennon, 2024-10-17:
Thanks for this @niraami. Could you raise a PR and we'll build with your changes in the CI to test?

niraami, 2024-10-17:
Sure thing, but I really don't know if it's the right solution, just something that worked for us :grin:

closed 2024-10-17 by ajlennon

zboszor, 2024-10-18:
I am late to the party but #239 looked good.

PR #239 (2024-10-17): Fix native libdir path for multilib
Closes #238 

When building for an image with multilib (for 32 bit x86), the `${libdir}` variable gets changed from `/usr/lib` to `/usr/lib64` - but `mono-native` still outputs the libraries into `/usr/lib`.
This is one approach that fixes that issue.

Note: this also should be backported as far back as Kirkstone, maybe further - I dunno what your workflow for that is.