DynamicDevices/meta-mono #257: [scarthgap] Seeing build failures for msbuild-native 16.6

awais11010, 2025-03-11:
Hi,

We're seeing issues while building msbuild-native on yocto scarthgap. My version configuration is as follows
```
PREFERRED_VERSION_mono = "6.12.0.161"
PREFERRED_VERSION_mono-native = "6.12.0.161"
PREFERRED_VERSION_msbuild = "16.6"
PREFERRED_VERSION_msbuild-native = "16.6"
```

and the build failure I have is
```
Microsoft (R) Build Engine version 16.3.0-ci for Mono
Copyright (C) Microsoft Corporation. All rights reserved.

Some command line switches were read from the auto-response file "MSBuild.rsp". To disable this file, use the "-noAutoResponse" switch.
/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/eng/../artifacts/mono-msbuild/MSBuild.dll /fl /bl /clp:Summary /flp:v=diag /m /p:MSBuildExtensionsPath=/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/eng/../artifacts/mono-msbuild/Extensions /p:MSBuildExtensionsPath32=/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/eng/../artifacts/mono-msbuild/Extensions /p:MSBuildExtensionsPath64=/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/eng/../artifacts/mono-msbuild/Extensions /p:DeterministicSourcePaths=false /p:MicrosoftNetCompilersVersion=3.6.0-4.20224.5 /p:NuGetBuildTasksVersion=5.6.0-rtm.6591 /verbosity:minimal mono/build/update_bundled_bits.proj
Build started 3/11/2025 9:44:39 AM.
Restoring package named Microsoft.NET.Sdk version='3.1.300-preview.20214.15'
/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/Directory.Build.props(3,3): warning : Unable to load the service index for source https://dotnetfeed.blob.core.windows.net/dotnet-core/index.json.
/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/Directory.Build.props(3,3): error : Unable to find package Microsoft.DotNet.Arcade.Sdk with version (= 1.0.0-beta.20202.9)
/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/Directory.Build.props(3,3): error :   - Found 1397 version(s) in arcade [ Nearest version: 1.0.0-beta.20213.4 ]
/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/Directory.Build.props(3,3): error :   - Found 178 version(s) in dotnet-tools [ Nearest version: 5.0.0-beta.19467.5 ]
/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/Directory.Build.props(3,3): error :   - Found 0 version(s) in nuget.org
/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/Directory.Build.props(3,3): error :   - Found 0 version(s) in dotnet-core
/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/Directory.Build.props(3,3): error :   - Found 0 version(s) in nuget-build
/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/Directory.Build.props(3,3): error : Version 8.0.303 of the .NET Core SDK requires at least version 17.8.3 of MSBuild. The current available version of MSBuild is 0.0.0.0. Change the .NET Core SDK specified in global.json to an older version that requires the MSBuild version currently available.
/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/Directory.Build.props(3,31): error MSB4236: The SDK 'Microsoft.DotNet.Arcade.Sdk' specified could not be found. [/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/mono/build/sdks_and_nugets/sdks/sdks.csproj]
Restoring package named Microsoft.NET.Sdk.Razor version='3.1.2'
/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/Directory.Build.props(3,31): error MSB4236: The SDK 'Microsoft.DotNet.Arcade.Sdk' specified could not be found. [/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/mono/build/sdks_and_nugets/sdks/sdks.csproj]
Restoring package named Microsoft.NET.Sdk.Web version='3.1.300-servicing.20216.7'
/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/Directory.Build.props(3,31): error MSB4236: The SDK 'Microsoft.DotNet.Arcade.Sdk' specified could not be found. [/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/mono/build/sdks_and_nugets/sdks/sdks.csproj]
Restoring package named Microsoft.NET.Sdk.Publish version='3.1.300-servicing.20216.7'
/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/Directory.Build.props(3,31): error MSB4236: The SDK 'Microsoft.DotNet.Arcade.Sdk' specified could not be found. [/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/mono/build/sdks_and_nugets/sdks/sdks.csproj]
Restoring package named Microsoft.NET.Sdk.Web.ProjectSystem version='3.1.300-servicing.20216.7'
/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/Directory.Build.props(3,31): error MSB4236: The SDK 'Microsoft.DotNet.Arcade.Sdk' specified could not be found. [/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/mono/build/sdks_and_nugets/sdks/sdks.csproj]
Restoring package named ILLink.Tasks version='0.1.6-prerelease.19380.1'
/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/Directory.Build.props(3,31): error MSB4236: The SDK 'Microsoft.DotNet.Arcade.Sdk' specified could not be found. [/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/mono/build/sdks_and_nugets/sdks/sdks.csproj]
Restoring package named NuGet.Build.Tasks.Pack version='5.6.0-rtm.6591'
/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/Directory.Build.props(3,31): error MSB4236: The SDK 'Microsoft.DotNet.Arcade.Sdk' specified could not be found. [/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/mono/build/sdks_and_nugets/sdks/sdks.csproj]
Restoring package named FSharp.NET.Sdk version='1.0.4-bundled-0100'
/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/Directory.Build.props(3,31): error MSB4236: The SDK 'Microsoft.DotNet.Arcade.Sdk' specified could not be found. [/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/mono/build/sdks_and_nugets/sdks/sdks.csproj]
Restoring package named Microsoft.NET.Build.Extensions version='3.1.300-preview.20214.15'
/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/Directory.Build.props(3,31): error MSB4236: The SDK 'Microsoft.DotNet.Arcade.Sdk' specified could not be found. [/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/mono/build/sdks_and_nugets/sdks/sdks.csproj]
Restoring package named Microsoft.DotNet.MSBuildSdkResolver version='3.1.300-preview.20227.1'
/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/Directory.Build.props(3,31): error MSB4236: The SDK 'Microsoft.DotNet.Arcade.Sdk' specified could not be found. [/home/awaisb/yocto_scarthgap/tmp/work/x86_64-linux/msbuild-native/16.6/git/mono/build/sdks_and_nugets/sdks/sdks.csproj]

Build FAILED.
```

Interestingly, I believe this has started happening recently as I was able to build the same configuration earlier and only saw this on a clean setup that was put together two days ago.

Any help or pointers will be highly appreciated.

CrazyGecko, 2025-03-11:
I had the same issue. From my understanding, Microsoft don't provide the package we use here anymore. This issue is not related to scarthgap, it's general.
Is it possible, that you cleaned the downloads folder between these two builds? Using version `16.10.1` did fix it for me.

CrazyGecko, 2025-03-13:
Update: because these packages are downloaded whilst the build, they lie in the `sstate-cache` cleaning the cache will remove the package and it can't be re-downloaded (not available anymore).

ajlennon, 2025-03-13:
Oh dear! OK well on another project I recently found I needed to host some packages and I was able to do this on GitHub So if somebody can get me the missing packages I will try to host them alongside this repo?

CrazyGecko, 2025-03-14:
I can try to take it out one of our `sstate` which still builds on Monday.
Possible, that the drop could create some troubles, as the application of @awais11010 does already not build on `16.10`
Does anyone know, if `15.4` still builds?

ajlennon, 2025-03-15:
I tell you what - let me just kick off a scarthgap build on the CI now and see what happens

CrazyGecko, 2025-03-15:
I fear it should work as it will take the newest msbuild recipe.

ajlennon, 2025-03-15:
Oh I am confused - I assumed it was using the breaking msbuild by default? Why would this be failing with the older msbuild if our default is the working one? I suspect I am getting the wrong end of the stick somewhere?

CrazyGecko, 2025-03-15:
I hope this short summary helps you:
The problem is only with the msbuild recipes and is not linked to scarthgap. The problem is with the recipe msbuild version `16.6`. When no version is declared, `16.10` is used.  This is why I asked also if someone have a problem with `15.4`. Maybe the title is misleading.

abelal, 2025-03-17:
Unless one already has a sstate/download cached locally

ajlennon, 2025-03-17:
Sounds good. Let's drop them then

CrazyGecko, 2025-03-19:
I also agree on dropping them 👍

ajlennon, 2025-03-19:
Have kicked off a bunch of CI builds...

ajlennon, 2025-03-21:
OK I think that's sorted out. I will close this issue but feel free to reopen it if I missed anything

closed 2025-03-21 by ajlennon