DynamicDevices/meta-mono #260: Fix DEPENDS of gtk-sharp libglade for glade

cleitonbueno, 2025-04-24:
In the scarthgap version there is no libglade recipe or libglade PACKAGE, but rather glade.

My proposal:

```
--- a/recipes-mono/gtk-sharp/gtk-sharp3_2.99.4.bb
+++ b/recipes-mono/gtk-sharp/gtk-sharp3_2.99.4.bb
@@ -2,7 +2,7 @@ require gtk-sharp.inc
 
 inherit pkgconfig
 
-DEPENDS += " gtk+3 atk pango cairo glib-2.0 libglade mono"
+DEPENDS += " gtk+3 atk pango cairo glib-2.0 glade mono"
 RDEPENDS:${PN} += " perl gtk+3"
 
 LIC_FILES_CHKSUM = "file://COPYING;md5=8754deb904d22254188cb67189b87f19"
```

ajlennon, 2025-04-25:
I haven't checked but given this is an issue and presumably the CI build didn't catch it can you add the gtk-sharp build to the CI image?

cleitonbueno, 2025-04-25:
How? Something like this in the `meta-mono/recipes-mono/images/test-image-mono.bb` image:

```
--- a/recipes-mono/images/test-image-mono.bb
+++ b/recipes-mono/images/test-image-mono.bb
@@ -8,6 +8,7 @@ DEFAULT_TEST_SUITES:pn-${PN}:prepend = "dotnet "
 IMAGE_INSTALL += "msbuild \
                   dotnet \
                   dotnet-helloworld \
+                  gtk-sharp3 \
                   python3-pythonnet \
 "

```

ajlennon, 2025-04-26:
Looks good yes