DynamicDevices/meta-mono #262: Update gtk-sharp3

cleitonbueno, 2025-04-28:
**Description:**

I propose updating the gtk-sharp3 from commit [9a72bb67fff7e4845b7bb430a608282668c3e4da](https://github.com/mono/gtk-sharp/commit/9a72bb67fff7e4845b7bb430a608282668c3e4da) to the latest commit [dadc19cf1b90c5743f2776c675faac990e397a56](https://github.com/mono/gtk-sharp/commit/dadc19cf1b90c5743f2776c675faac990e397a56).

This update brings important compatibility improvements and build fixes.

**Key changes included:**

- Change default branch from master to main.
Commit [a00552ad](https://github.com/mono/gtk-sharp/commit/a00552ad)

- Fix compilation error:
CS0030: Cannot convert type 'Gtk.VScale' to 'System.Range'
Commit [a00552ad](https://github.com/mono/gtk-sharp/commit/a00552ad)

- Fix build issue with newer Mono versions, fix this Issue [#151](https://github.com/DynamicDevices/meta-mono/issues/151)
`MONO_PROFILE_ENTER_LEAVE` undeclared
Commit [c1b9bd42](https://github.com/mono/gtk-sharp/commit/c1b9bd42)

- Replace usage of mcs with csc for building, and remove the patch (0001-fixup-gmcs-to-mcs.patch) that does this.
Commit [05e47a49](https://github.com/mono/gtk-sharp/commit/05e47a49)

- Add gtk-sharp3 in `test-image-mono` image.

ajlennon, 2025-04-28:
Sounds good!

ajlennon, 2026-02-09:
Thanks for the pointer - will get this going too